TheImagingSource/IC3.5-Imaging-Control-Samples #3: Runtime error with Python example

eehyc, 2019-04-22:
Hello, 

I keep getting a runtime error while running the Python sample code (tis-OpenCV.py) in the line  Camera.SetPropertyValue("Gain","Value",10):

Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.1.1\helpers\pydev\_pydev_bundle\pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.1.1\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "E:/XXX-alpha/code/dynamic/IC-TIS/tis-OpenCV.py", line 79, in <module>
    Camera.SetPropertyValue("Gain","Value",10)
  File "E:\XXX-alpha\code\dynamic\IC-TIS\tisgrabber.py", line 618, in SetPropertyValue
    Value)
OSError: [WinError -529697949] Windows Error 0xe06d7363

The platform is Windows 10, Python 3.6 with DFK 31BU03.H. Do you know how to solve the issue?

Thanks

TIS-Stefan, 2019-04-23:
Hello

The error is caused by the gain range: It is 260 - 1023. Therefore, the value 10 will not fit.

tcam-ctrl -p serialnumber shows the properities and their ranges.

Stefan

TIS-Stefan, 2019-04-23:
Ups, That is Windows
You can use IC Capture to check the available property ranges.

Stefan

eehyc, 2019-04-24:
Thank you for your prompt reply !

Best

closed 2019-04-24 by eehyc