TheImagingSource/IC3.5-Imaging-Control-Samples #4: Capture 16 bit greyscale imaging

jonasze, 2019-04-26:
The SnapImage() function seems to be hardcoded to return 8bit images, even if run in Y16 Video Format. Is there a way to grab a 16bit image in Python?

TIS-Stefan, 2019-04-26:
Yes:
SetFormat(SinkFormats.Y16) should do the job.
Stefan

jonasze, 2019-04-26:
Many thanks for the quick reply! This (probably) worked, it changes the image grabbed to (width, height,**2**) instead of (width, height,**3**) for the RGB24 case. 

Still the dtype is uint8. Do I have "sum" the two layers to get the 16bit image?

TIS-Stefan, 2019-04-26:
hmm

to be tested. Currently I work on the property interface, so you may wait until next week, However, did you check my Python code in tisgrabber.py, whether I did the correct things?

The sinkformat must be set, before the live video is started. Did you do so?

Stefan

closed 2019-04-26 by TIS-Stefan

jonasze, 2019-04-26:
Sure, next week is completely fine!
From the code in tisgrabber.py I indeed understand that two 8bit images are stacked, but my knowledge of interfacing python and C is not very good....

TIS-Stefan, 2019-04-26:
I got my today's task nearly done, Thus I will have a look on Monday. 

I hope, that is fine for you.

Stefan

eehyc, 2019-04-27:
I have a question related to this issue. How about 12bit RGB image?

 Thanks!

TIS-Stefan, 2019-04-29:
Hello

I updated the repository at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/tree/master/Python. There is a new sample "Using-Y16.py" and an updated "tisgrabber.py"

RGB64 for 12bit / 10bit RGB is currently not supported. Use Y16 instead, however, that is not debayered.

I will add RGB64 later.

(I must admit, image buffer handling in the tisgrabber.dll is quite primitive. There is space for enhancement.)

Stefan

jonasze, 2019-04-30:
Dear Stefan,

many thanks for the quick update! It seems to work and returns now a 2D array with uint16 as desired but I struggle to make sense of the data. The 2D array in the image variable:
`
  image = Camera.GetImageEx()`
is not arranged how it should be. I focused a leaser beam in the middle of the image and would expect lots of saturated values in the middle of the array (as seen in IC capture), but instead these saturated pixels are smeared out over many lines. 
-> It seems there is some transformation that is not correct, e.g. how from a long 1D array the 2D array is formed.

TIS-Stefan, 2019-04-30:
Hello Jonas

Could you please check for me, whether I made the numpy stuff correct in the tisgrabber.py? I am not an experiecend Python programmer, therefore, I am not sure.
The array should have two dimensions width and height of uint 16.

Thank you in advance!

Stefan

jonasze, 2019-04-30:
Hi Stefan,

nevermind it works well, I just forgot the cv.flip and cv.erode transformations that you are doing in tis-openCV...

Many thanks!
Jonas

TIS-Stefan, 2019-04-30:
Hi Jonas

thank you! That is a relief for me,

Stefan

jonasze, 2019-05-03:
Hi Stefan,

one more comment: now the 12bit image is scaled over a 16bit array. The values are then 16 , 32 ,6 4 and so on until 65535. Is this as intended or should it be 1 , 2 ,3 etc until 4096?

For me it makes no difference, but maybe for others.

best
Jonas

TIS-Stefan, 2019-05-03:
Hello Jonas

we use the upper 12 bits only. The lower 4 bits are not connected. Therefore, white is 65535, black is 0-15. The lower 4 bits are invalid.
You get a range from 0 to 4096, if you shift 4 bits to the right " >> 4"

Stefan

jonasze, 2019-05-03:
Ok, I did it by using 
`image=np.right_shift(image,4)`
cheers
Jonas

Spexophis, 2023-05-22:
> Hello
>
> I updated the repository at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/tree/master/Python. There is a new sample "Using-Y16.py" and an updated "tisgrabber.py"
>
> RGB64 for [...]

Hi, Is the sample about Y16 still available?

TIS-Stefan, 2023-05-22:
Hello 

The sample exists:
[Using-Y16.zip](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/11531825/Using-Y16.zip)
I did not test it. 

Stefan

Spexophis, 2023-05-23:
> Hello
> 
> The sample exists: [Using-Y16.zip](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/11531825/Using-Y16.zip) I did not test it.
> 
> Stefan

Thank you for sharing this! 
My question is basically how to interpret the output of the Y16 image, which is two 2d array.
Does each 2d array contains 8 bits of the 16 bits for each pixel? I tried to multiplied one of the array by 2**8 and add them up, the result looks not bad, but not as good as the result saved from IC Capture software. I am wondering if I did it correctly.
btw: according to this website, it seems all the bits have to be used to get the final pixel values.
https://www.theimagingsource.com/en-us/documentation/icimagingcontrolcsharp/PixelformatY16.htm

TIS-Stefan, 2023-05-23:
The 2D Array should contain ushort values, which is a two byte unsigned integer. It contains the 16 bits. Depending on the used camera model, the upper 12 or the upper 10 bits contain valid image data. How do you display the result?

Spexophis, 2023-05-23:
`    def get_data(self):
        buffer_size, width, height, depth = self.get_buffer()
        image_pointer = ic.IC_GetImagePtr(self.hGrabber)
        image_data = ctypes.cast(image_pointer, ctypes.POINTER(ctypes.c_ubyte * int(buffer_size)))
        image = np.array(image_data.contents)
        image = np.reshape(image, (int(height.value), int(width.value), depth))
        return image[:, :, 0] + image[:, :, 1] * (2 ** 8)`

This is what I do on the data. Please correct me if anything is wrong. Thank you!

TIS-Stefan, 2023-05-23:
I am not sure what the contents of depth is. It should be 16, because you have 16 bits. Please check this. 
The correct usage is:
```Python
              # Declare variables of image description
                Width = ctypes.c_long()
                Height = ctypes.c_long()
                BitsPerPixel = ctypes.c_int()
                colorformat = ctypes.c_int()

                # Query the values of image description
                ic.IC_GetImageDescription(hGrabber, Width, Height,
                                          BitsPerPixel, colorformat)

                # Calculate the buffer size, get the number of bytes per pixel
                # and the data type in the numpy array.
                elementsperpixel = 1
                dtype = np.uint8
                if colorformat.value == tis.SinkFormats.Y800:
                    elementsperpixel = 1  # 1 byte per pixel
                if colorformat.value == tis.SinkFormats.Y16:
                    dtype = np.uint16
                    elementsperpixel = 1  # 1 uint16 per pixel
                if colorformat.value == tis.SinkFormats.RGB24:
                    elementsperpixel = 3  # BGR format, 3 bytes
                if colorformat.value == tis.SinkFormats.RGB32:
                    elementsperpixel = 4  # BGRA format, 4 bytes

                buffer_size = Width.value * Height.value * int(float(BitsPerPixel.value) / 8.0)

                # Get the image data
                imagePtr = ic.IC_GetImagePtr(hGrabber)

                imagedata = ctypes.cast(imagePtr,
                                        ctypes.POINTER(ctypes.c_ubyte *
                                                       buffer_size))

                # Create the numpy array
                image = np.ndarray(buffer=imagedata.contents,
                                   dtype=dtype,
                                   shape=(Height.value,
                                          Width.value,
                                          elementsperpixel))
```
Surprisingly I found this sample: 
[12-image-processing-16bit.zip](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/11542981/12-image-processing-16bit.zip)

Stefan

Spexophis, 2023-05-23:
> I am not sure what the contents of depth is. It should be 16, because you have 16 bits. Please check this. The correct usage is:
>
> ```python
>               # Declare variables of image [...]

Thank you for the finding! I finally realize why the images saved from python are not as good as the images saved from the software. It is due to the noise reduction. I set the noise reduction to 8 frames in the IC Capture, but the python code I only set the denoise property. I will just need to figure out how to set the noise reduction function then it should be the same quality.

TIS-Stefan, 2023-05-23:
IC Capture makes averaging of images on its own, therefore, it is not a camera nor an IC Imaging Control property.  You can use OpenCV for that with cv2.accumulateWeighted()

```Python
        # Convert 16bit image to 8 bit image
        if img.dtype == np.dtype('uint16'):
            self.display_image = cv2.convertScaleAbs(img, alpha=(1.0/256.0))
        else:
            self.display_image = img

        if self.avg is None:
            self.avg= np.float32(img)
   
        cv2.accumulateWeighted(self.display_image, self.avg, 0.075)
```
I am not sure, why I convert the 16 bit ot 8 bits and whether that is necessary. This copied from a motion detection sample I found in the depth of my hard disc.

Stefan

Spexophis, 2023-05-23:
> IC Capture makes averaging of images on its own, therefore, it is not a camera nor an IC Imaging Control property. You can use OpenCV for that with cv2.accumulateWeighted()
>
> ```python
>         # [...]

I see. Then I can just take multiple frames and average them in python. I think the problem is solved! Thank you very much.

TIS-Stefan, 2023-05-24:
Consider using OpenCV, because it is fast than programming the same stuff in Python. 

Stefan