TheImagingSource/IC3.5-Imaging-Control-Samples #6: Python Expsoure times limited

jonasze, 2019-05-10:
I am using a DMK UX183 which supports exposure times from 1us to 30s which works well in IC Capture. But when using python, more concretely
`Camera.SetPropertyAbsoluteValue("Exposure","Value",2)` to set e.g. a 2s exposure time the image returned is only 0s.
Same for exposure times <0.001.

TIS-Stefan, 2019-05-13:
Dear Mr. 

Thank you for your email. Please try the code below:

Camera.SetPropertySwitch("Exposure","Auto",0)
Camera.SetPropertyAbsoluteValue("Exposure","Value",2) 
ExposureTime=[0]
Camera.GetPropertyAbsoluteValue("Exposure","Value",ExposureTime)
print("Exposure time abs: ", ExposureTime[0])

I used that with Python3.6 and it works as expected. The update of the live images is every two seconds.

Yours sincerely / Mit freundlichen Grüßen 

Stefan Geissler / Stefan Geißler 

-------------------------------------------------------------------- 
The Imaging Source Europe GmbH 
Überseetor 18 
D-28217 Bremen 
Germany 
Tel   *49/421/33591-0 
Fax   *49/421/33591-80 
Registergericht: Amtsgericht Bremen HRB 11 927 
Geschäftsführer: Rolf Bollhorst  

-------------------------------------------------------------------- 
 Was this message helpful?   Yes   [https://theimagingsource.deskpro.com/ticket-rate/WZBZ-2642-WIVB/DKS8TSHKADXQT9N/43395?setrating=1]   It was OK   [https://theimagingsource.deskpro.com/ticket-rate/WZBZ-2642-WIVB/DKS8TSHKADXQT9N/43395?setrating=0]   No   [https://theimagingsource.deskpro.com/ticket-rate/WZBZ-2642-WIVB/DKS8TSHKADXQT9N/43395?setrating=-1] 

View and manage this ticket online: https://theimagingsource.deskpro.com/tickets/9663 

  The Imaging Source [https://www.theimagingsource.com] · https://theimagingsource.deskpro.com/ 

 (#OHRDKS8TSHKADXQT9N)

jonasze, 2019-05-14:
Hi Stefan,

I did but the error persists. I use the following code
`import ctypes as C
import tisgrabber as IC
import cv2
import numpy as np
import matplotlib.pyplot as plt

Camera = IC.TIS_CAM()

Devices = Camera.GetDevices()
for i in range(len( Devices )):
    print( str(i) + " : " + str(Devices[i]))

Camera.open("DMK 33UX183 16910441")

Camera.SetVideoFormat('Y16 (5472x3648)')
Camera.SetFormat(IC.SinkFormats.Y16)
Camera.StartLive(0)    

Camera.SetPropertySwitch("Exposure","Auto",0)
Camera.SetPropertyAbsoluteValue("Exposure","Value",2) 
ExposureTime=[0]
Camera.GetPropertyAbsoluteValue("Exposure","Value",ExposureTime)
print("Exposure time abs: ", ExposureTime[0])

Gainauto=[0]
Camera.GetPropertySwitch("Gain","Auto",Gainauto)

Camera.SetPropertySwitch("Gain","Auto",0)
Camera.SetPropertyValue("Gain","Value",0)

Camera.SnapImage()
image = Camera.GetImageEx()
image = cv2.flip(image,0)
image = cv2.erode(image,np.ones((11, 11)))
image=np.right_shift(image,4)

Camera.StopLive()   `

For e.g. exposure value 2 the variable image only contains zeros while for smaller integration times the image is as expected. I tested also small exposure times again and this time they work also for 1/20000.

TIS-Stefan, 2019-05-14:
I suppose, the Camera.SnapImage() returns to early or with an image taken to early. Call this twice and see, what happens.

Stefan

jonasze, 2019-05-14:
Your are correct and now build a while loop that circumvents the problem:

> image=0
while np.max(image)==0:
    Camera.SnapImage()
    image = Camera.GetImageEx()

TIS-Stefan, 2019-05-14:
The reason is the camera provides an image, before the new set exposure time comes in effect. That is the black one you see.

closed 2021-02-19 by TIS-Stefan