TheImagingSource/IC3.5-Imaging-Control-Samples #7: Camera.close() missing leading to memory leak

jonasze, 2019-05-15:
When constructing the Camera object in some subfunction and grabbing images there, Python will not garbage collect the function since no Camera.close() was ever performed. This rapidly leads to an OutOfMemory Error in Pyton.

TIS-Stefan, 2019-05-16:
If a camera object is created, of course memory is allocated. This must be released again. Ususlayy IC_RealeaseGrabber (that is the call into the library) does the job. I am currently not show, how this is implemented in the Python wrapper.

wwoodgate, 2019-06-13:
Hello, we have the same issue preventing operationalising Imaging Source cameras with python, which was a reason why we selected Imaging Source. There is no deallocate method of the Camera handle in the python wrapper. Is it possible to implement this option? Many thanks

TIS-Stefan, 2019-06-13:
Can you show me, what your source is doing?
Do have an idea, where the memory is accumulated?
I am not that Python expert, but I will try to figure out what happens with your help.

Stefan

wwoodgate, 2019-06-13:
Thanks Stefan. Unfortunately python does not give any meaningful errors. In some instances the snapped image will be empty, and in other instances the Camera.GetDevices() object will not find any camera when trying to reload the Camera. Only a power cycle and kernel restart seems to fix the issue in those circumstances.

TIS-Stefan, 2019-06-13:
Hi

I thought, we speak about a memory leak...
You may check, whether IC Capture lists the camera, when the Python does not do. Both use the same IC Imaging Control DLLs.

Stefan

wwoodgate, 2019-06-17:
Hi Stefan, the original problem remains - there appears to be no camera close function in either python or the IC Imaging Control v3.4 Demo App.

When I operate the camera in the IC software and close it I cannot then operate the camera in a python environment without unplugging and re-plugging the camera's USB connection. I have the same issue if I first open the camera in python and then try to open it in the IC software. If functionality was added so that the driver was released to create a clean camera exit that may solve the issue.

wwoodgate, 2019-06-18:
in dll header - shows clean up procedure.

`
HGRABBER hGrabber;
if( IC_InitLibrary(0) == IC_SUCCESS )
{
	hGrabber = IC_CreateGrabber();
	if( hGrabber )
	{
		if( IC_OpenVideoCaptureDevice(hGrabber,"DFK 21F04") == IC_SUCCESS )
		{
			// .. do something with the video capture device.
			// Now clean up.
			IC_CloseVideoCaptureDevice( hGrabber );
			IC_ReleaseGrabber( hGrabber );
		}
	        IC_CloseLibrary();
        }
}`

-
IC_CloseVideoCaptureDevice,
IC_ReleaseGrabber,
IC_CloseLibrary
-
are not implemented in the python wrapper.

TIS-Stefan, 2019-07-08:
Hi,

You are right, not all functions are implemented. Did you do that on your own?

Stefan

mik-p, 2019-07-11:
Had a go at adding to python wrapper but couldn't get it working. any chance of this being implemented? without cleaning up resources apps with python wrapper are quite brittle

TIS-Stefan, 2019-07-11:
Hello 

I implemented the functions, but I am not sure about ReleaseGrabber(). I must admit, I am not a Python expert.
So you have now:
```
import tisgrabber as IC

IC.TIS_GrabberDLL.InitLibrary(None)
Camera = IC.TIS_CAM()
Camera.open("DFK 38UX304 43619905")
Camera.StartLive(1)
Camera.StopLive()
Camera.ReleaseGrabber()
IC.TIS_GrabberDLL.CloseLibrary()
```
[tisgrabber.zip](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/3382582/tisgrabber.zip)

I hope, this helps.

Stefan

wwoodgate, 2019-07-12:
Hi Stefan, unfortunately ReleaseGrabber() does not work. Any other ideas?

TIS-Stefan, 2019-07-12:
Hi
Please specify "does not work".
On my system I got no error.

Stefan

TIS-Stefan, 2019-07-12:
BTW:
Can you please give me a small sample script, that shows the memory issue?

Thanks.

closed 2021-02-19 by TIS-Stefan