TheImagingSource/IC3.5-Imaging-Control-Samples #8: Continuous AE

bgillett-acertaralabs, 2019-05-23:
Hello,

I'm working with a GigE camera and the _tisgrabber.py_ library on Windows in Python2.7. I'm trying to configure it such that the Auto Exposure system is continuously running, so when I snap an image, it will already have been auto-exposed. To provide a specific example, here's a little code:

`
import cv2
import tisgrabber as IC

# Set up camera
Camera = IC.TIS_CAM()
Camera.open('DFM 25GP031-ML 00000000') # insert real serial number...
Camera.SetVideoFormat('RGB32 (640x480)')
Camera.SetFrameRate(8)
Camera.SetPropertySwitch('Exposure', 'Auto', 1)
Camera.SetPropertySwitch('Gain', 'Auto', 1)
Camera.SetPropertySwitch('WhiteBalance', 'Auto', 1)
Camera.SetPropertyValue('Exposure', 'Auto Reference', 128)
Camera.StartLive(0)

# take first image
Camera.SnapImage()
image = Camera.GetImage()
# display it
cv2.imwrite('image1.png', image)
cv2.imshow('Image', image)
cv2.waitKey(0)

# now, scene has changed: AE needs to have readjusted

# take second image
Camera.SnapImage()
image = Camera.GetImage()
# display it
cv2.imwrite('image2.png', image)
cv2.imshow('Image', image)
cv2.waitKey(0)

cv2.destroyWindow('Image')
Camera.StopLive()
`
I test this code by
1) pointing the camera at my desk (indoor lighting),
2) viewing the image and pointing the camera out the window (bring outdoor lighting) while the program waits,
3) pressing a key and viewing the new image.

While the code runs fine, the second image is never exposed correctly. I've attached image1.png and image2.png for reference.
I'd like the AE to "keep up" even when I'm not actively snapping images. How could I achieve that?

Any help would be much appreciated. Thanks!
[image]
[image]

TIS-Stefan, 2019-05-24:
Hello

Your code looks fine to me and the camera / driver should react accordingly. 
I would like you to try IC Capture and see,  how the camera behaves in the same situation.  IC Capture can be downloaded from https://www.theimagingsource.com/support/downloads-for-windows/end-user-software/iccapture/

Also I would like you to send me image2.png. I will check the average brightness. If the average brightness is around 128, then the AE works correctly.

Stefan

bgillett-acertaralabs, 2019-05-24:
Hi --  thanks for the reply!

When I do the same test in IC Capture, it exposes correctly within a few seconds as desired.
Image2.png is actually attached above, it's the completely white image below the first one of the keyboard. The camera didn't auto-expose when I pointed it outside, so the image it captured was completely washed out.

Thanks!

bgillett-acertaralabs, 2019-05-24:
Update:

Looks like ```Camera.SetContinuousMode(0)``` used with a callback as seen in _callback.py_ did the trick for me. Now, the camera is continuously updating its AE.
Thanks again.

closed 2019-05-24 by bgillett-acertaralabs