TheImagingSource/IC3.5-Imaging-Control-Samples #9: setPropertySwitch for Exposure Auto

bgillett-acertaralabs, 2019-05-24:
Hello,

I'm using _tisgrabber.py_ on Windows with a GigE camera. I'm trying to turn off AE on the camera temporarily. I have been using calls such as ```setPropertySwitch('Gain', 'Auto', 0)```. With Gain and WhiteBalance, I haven't had any difficulty. However, ```setPropertySwitch('Exposure', 'Auto', 0)``` doesn't work as desired. For example:

```
import time
import tisgrabber as IC

# Set up camera
Camera = IC.TIS_CAM()
Camera.open('DFM 25GP031-ML 00000000') # insert real serial number...
Camera.SetVideoFormat('RGB32 (640x480)')
Camera.SetFrameRate(8)
Camera.StartLive(0)

Camera.SetPropertySwitch('Gain', 'Auto', 0)
print('Set Auto Exposure Returned: ' + str(Camera.SetPropertySwitch('Exposure', 'Auto', 0)))
Camera.SetPropertySwitch('WhiteBalance', 'Auto', 0)

# Set some values
print('Set Gain Returned: ' + str(Camera.SetPropertyValue('Gain', 'Value', 6)))
print('Set Exposure Returned: ' + str(Camera.SetPropertyValue('Exposure', 'Value', 1234)))

# query them
print('\nImmediately after configuring:')
print('Gain: ' + str(Camera.GetPropertyValue('Gain', 'Value')))
print('Exposure: ' + str(Camera.GetPropertyValue('Exposure', 'Value')))

# wait a bit
time.sleep(0.5)

# query again
print('\nAfter waiting:')
print('Gain: ' + str(Camera.GetPropertyValue('Gain', 'Value')))
print('Exposure: ' + str(Camera.GetPropertyValue('Exposure', 'Value')))

Camera.StopLive()

```

This code produces the following results on my system:

```
Set Gain Auto returned: 1
Set Exposure Auto returned: 1
Set WhiteBalance Auto returned: 1
Set Gain Value returned: 1
Set Exposure Value returned: 1

Values immediately after configuring:
Gain: 6
Exposure: 1234

Values after waiting:
Gain: 6
Exposure: 3410 <or some other value not equal to 1234>
```

I would expect the values for Exposure to remain the same, as I did not change them and AE should be turned off and not changing them either. However, the camera behaves as though 'Auto Exposure' was not actually disabled. I cross checked this by examining the _Property Page_ for the device before and after running the program:
[image]

As seen in the screenshot, 'Auto Max Exposure' was turned off, but not 'Auto Exposure'. Here is a screenshot from the _VCD Property Inspector_ of the properties of 'Exposure' on this device:

[image]

There are _two_ properties by the name 'Auto' under 'Exposure'. Perhaps this is the cause of my difficulty, as there may not be a way to distinguish between the two, and the 'Auto Max Exposure' property is taking precedence for arbitrary reasons?

Perhaps my understanding is incorrect and there's a solution I have missed.
Your advice is appreciated -- I'd like to figure out how to programmatically turn off Auto Exposure and instead set it manually.

My apologies for the long post, and thanks for your support.

~Ben

TIS-Stefan, 2019-05-27:
Hi Ben

You are right, there are two of them...  thus, for exposure please use
`IC_EnableAutoCameraProperty(hGrabber, PROP_CAM_EXPOSURE, 0); // disable autoexposure`
That should solve the issue.
I will fix that issue in the next tisgrabber.dll by giving a different name to the auto max exposure.

Stefan

bgillett-acertaralabs, 2019-06-05:
Stefan,

Sorry for the delayed reply. Your solution worked, thank you very much!

Ben

thladnik, 2020-03-20:
Hi @TIS-Stefan, 

I have the exact same issue with the exposure setting in Python. `Camera.SetPropertySwitch('Exposure', 'Auto', 0)` does not work. Where am I supposed to use `IC_EnableAutoCameraProperty(hGrabber, PROP_CAM_EXPOSURE, 0);`? 

Tim

TIS-Stefan, 2020-03-20:
>  Where am I supposed to use IC_EnableAutoCameraProperty(hGrabber, PROP_CAM_EXPOSURE, 0);?
You are suppose to write that on a position in your code, on which you want to disable the automatic of Exposure.

Stefan

thladnik, 2020-03-20:
> 
> 
> > Where am I supposed to use IC_EnableAutoCameraProperty(hGrabber, PROP_CAM_EXPOSURE, 0);?
> > You are suppose to write that on a position in your code, on which you want to disable the automatic of Exposure.
> > Ok sorry, that is not that helpful. I would place the call after you have opened the camera.
> 
> Stefan

Thanks for your reply, but `IC_EnableAutoCameraProperty` doesn't appear to be a function implemented in tisgrabber.py

TIS-Stefan, 2020-03-20:
Hello

You may implement that on your own, that should not be too hard. Would that be ok for you?

Stefan

thladnik, 2020-03-23:
Hello Stefan, 

I've tried implementing it for a couple of hours now, without success. Could you please give me some pointers as to how to do this? 
Am I correct to assume that I first have to create a reference to the imported method `IC_EnableAutoCameraProperty` in the TIS_GrabberDLL class, 
like `SetAutoExposureSwitch = __tisgrabber.IC_EnableAutoCameraProperty`
and then set the `restype` and `argtypes` on `SetAutoExposureSwitch` and call it in a newly created method in TIS_CAM like so `error = TIS_GrabberDLL.SetAutoExposureSwitch(...)`?

If so then I coudn't get it to work. Where do I for example get `PROP_CAM_EXPOSURE` from?

Tim

TIS-Stefan, 2020-03-23:
Hello Tim

Do you really depend on Python?
However. 
Do you need to disable the automatic once and never enable it? If yes, then you may use the property dialog to disable the automatic and save all setting in a DeviceState file, as shown in the samples.
You can load this at program start and you are done. 

(I write this in this way, because I am at home and have not all possibilities as in my office)

Is this a way?

Stefan

thladnik, 2020-03-23:
Hello Stefan, 

unfortunately the whole application I'm using this for is written in Python, so in the long run having this as part of the Python bindings would be helpful. 
But thank you for this tip! Setting the property manually works well and is definitely good enough for development purposes right now. 

Would it be possible to include the auto exposure setting in some future release, when you get the chance? 

Tim

TIS-Stefan, 2020-03-23:
Hi Tim

I will try to download and check by tomorrow.

Stefan

TIS-Stefan, 2020-03-24:
Hi Tim

C to Python drives my crazy....

I updated the tisgrabber.py with two new functions. They are used as

```
    Camera.enableVideoAutoProperty(9,0) # Gain Auto off
    Camera.enableCameraAutoProperty(4,0) # Gain Auto off
```
Do not be shocked by magic numbers. They are from TISGrabberGlobalDefs.h file.
I hope, this helps.

thladnik, 2020-03-24:
Hi Stefan, 

thanks, this definitely helps. It works now! Sorry for the inconvenience...

Tim

TIS-Stefan, 2020-03-24:
Hi Tim

no need to excuse. It was my error and I had to find a solution. 

Stefan

closed 2020-03-24 by TIS-Stefan