TheImagingSource/IC3.5-Imaging-Control-Samples #10: in __init__     self._handle = _dlopen(self._name, mode) OSError: [WinError 126] 找不到指定的模块。

papicheng, 2019-06-05:
when I run the tis-Opencv using pycharm on windows, it shows that "in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] 找不到指定的模块。" cannot find the module.
I looked for the solution of this problem, but nothing helpful for me. 
If someone could give me some advise, I will be very appreciated.

TIS-Stefan, 2019-06-05:
Hello

I think, it is general Python issue you have there. It may also can be a Pycharm issue. You may try to find out, what self._name contains.
Can you run the program from command line, without PyCharm? 

BTW: I use Visual Studio Code for Python editing.

Stefan

papicheng, 2019-06-05:
aha, running it from command line works, it's odd to use PyCharm. Thanks for your reply！

closed 2019-06-06 by TIS-Stefan