TheImagingSource/IC3.5-Imaging-Control-Samples #12: Get property nama

julienGautier77, 2019-10-03:
Hi 
I'm using tissgraber.py  on python 3
I want to know if it is possible  to have all the property name available for one camer. Especially I need to have the range of Exposure and gain available (like getRangeMin('Exposure')
Thank to help me

TIS-Stefan, 2019-10-07:
Hello

I think, the sample in https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/01-list-properties.py is exactly, what you need.

Stefan

julienGautier77, 2019-10-07:
Hello
Thanks for your answer but I use a window sytem and i can not use Gi and gst 
I only use this repesitorie : https://github.com/TheImagingSource/IC-Imaging-Control-Samples/tree/master/Python
In particulary the code tisgrabber.py

Thanks in advance

TIS-Stefan, 2019-10-08:
Ah

now I understand. I mixed up the Linux and Windows repositories. 
I am very sorry, but the Python port is not as sophisticated as the IC Imaging Control .NET and Classlibrary SDK. Therefore, there is no function for listing all camera properties in. 
I would like to suggest to install the IC Imaging Control SDK from https://www.theimagingsource.com/support/downloads-for-windows/software-development-kits-sdks/icimagingcontrol/. It installs the "VCD Property Inspector" and provides a list of all properties. However, you can only copy the property names from there, but you can not enumerate the properties dynamically.

I would have to implement such functionality in the tisgrabber.dll first and port it to Python.

Stefan

julienGautier77, 2019-10-08:
Thanks 
If finnally find a solution in the tisgrabber.py program I add
in the TIS\cam class  : 
def GetPropertyRange(self,Property,Element):
            
            ValueMax = C.c_float()
            ValueMin = C.c_float()
            error = TIS_GrabberDLL.GetPropertyRange(self._handle,
                                                    self.s(Property),self.s(Element),
                                                   ValueMin,ValueMax)
            
            return (ValueMin.value,ValueMax.value)

and in the TIS_GrabberDLL class 
GetPropertyRange=__tisgrabber.IC_GetPropertyAbsoluteValueRange
    GetPropertyRange.restype= C.c_int
    GetPropertyRange.argtypes = (GrabberHandlePtr,
                                                  C.c_char_p,
                                                  C.c_char_p,
                                                  C.POINTER(C.c_float),
                                                  C.POINTER(C.c_float))

TIS-Stefan, 2019-10-08:
Perfect!

Stefan