TheImagingSource/IC3.5-Imaging-Control-Samples #13: Line 725 in tisgrabber.py has an unresolved reference at IC.ImageFileTypes

somtooo, 2019-10-18:
return TIS_GrabberDLL.SaveImage(self._handle, self.s(FileName),IC.ImageFileTypes[self.s(FileType)],Quality)

TIS-Stefan, 2019-10-21:
Aha.
Do you have some more context, e.g. a sample code for me, so I can reproduce that? That will be helpful.

somtooo, 2019-10-22:
Hello, 
Thanks for the reply. Forgot to mention im working with your python code. Secondly i didnt get the error by running the code instead I am getting a can not load dll file error.

 I just pulled from github and figured out my IDE had an error notification.
[image]. To reproduce error just pull code. Go to tisgrabber.py line 725.

 But this might just be because i use an ide i doubt a code editor will pick that up

closed 2019-10-28 by somtooo