TheImagingSource/IC3.5-Imaging-Control-Samples #14: Trying to load tisgrabber.dll gives me windows error

somtooo, 2019-10-18:
Ive tried editing the path- C:\\Users\\aaron\\PycharmProjects\\untitled\\tisgrabber_x64.dll
Still didnt work. Im i meant to do anything before trying to import. Thanks

WindowsError: [Error 126] The specified module could not be found

TIS-Stefan, 2019-10-21:
The TISUDSHL11_64.dll must by copied too.

Stefan

somtooo, 2019-10-22:
Hello, 
Thanks for the reply, I forgot to say im working with the python code. Also all dll files are in my directory. Find attached a picture of my directory
[image]

somtooo, 2019-10-22:
For more info 
This is code i ran which produced the error
[image]

This is the error
[image]

This is where the dll file gets loaded.
[image]

TIS-Stefan, 2019-10-23:
Hello
I am clueless. It work on all other computers, thus there must be a different issue. 
Your working directory is "c:\users\aaron\pycharmproject\untitled"? I do not know, how PyCharm sets the working directory. (I do not use PyCharm.)
You may open a command line window (cmd.exe), change into that directory and start the script:

```
>cd c:\users\aaron\pycharmproject\untitled
>pyhton webcamtest.py
```
Annotation: That will work with The Imaging Source cameras only.

Stefan

somtooo, 2019-10-23:
Hello,
Thanks  for the reply. Which text editor do you use. And yes im working on a project for Dwyer Instruments. The company recently purchased an Imaging Source camera

TIS-Stefan, 2019-10-24:
Hello

I use Visual Studio Code in Linux and Windows. However, Pycharm should work too, it is, from my point of view only a matter, whether the DLLs are in the working directory or in the windows search path. So you may set the working directory in the PyCharm project settings, if there is a setting like that.

Have you been able to run your Python script from command line, as I suggested? What happened?

Stefan

somtooo, 2019-10-25:
Hello good day,
I got this when i ran from cmd
[image]

somtooo, 2019-10-25:
But then i choose device name and it closes

TIS-Stefan, 2019-10-28:
Which camera model do you use? The library works with The Imaging Source cameras only.

Did you try IC Capture to, in order to make sure your camera works fine? It can be downloaded from https://www.theimagingsource.com/support/downloads-for-windows/end-user-software/iccapture/

Stefan

somtooo, 2019-10-28:
The device name is an image source camera. And i already tested the camera with the software. Works fine with very good quality.

TIS-Stefan, 2019-10-29:
And if you click on the combo box to get its items, nothing is shown?
Which software did you try?
What is the exact model name of your camera?

somtooo, 2019-11-01:
Hello,
This is the combo box,
[image]

This is the software i use 
[image]

Software works with camera very well just live video wont load with when i click ok on combo box

somtooo, 2019-11-01:
am i meant to install any other sdk to my system?

somtooo, 2019-11-01:
for more context here is a video of what happens

[image]

TIS-Stefan, 2019-11-04:
You program ends immediately after you started the live video. Therefore, the video ends immediately. You may add a while loop for keypress after you started the live video.

Stefan

somtooo, 2019-11-04:
Hey,
I made it hold by doing this and was able to take a picture
[image]
And here is it
[image]
in the ic capture software i can zoom out to see the gauge perfectly. Is it possible to do the same here because moving the camera all the way back isnt an option for now and i need an image of the gauge. 
The camera is currently 15inches from the gauge. Will changing the lens work?

closed 2021-02-19 by TIS-Stefan