TheImagingSource/IC3.5-Imaging-Control-Samples #15: IoT Application - XAML

emejoti, 2019-12-02:
Hello, I'm trying to generate and IoT app and use an imaging source camera in it.

I'm using the TIS.Imaging.ICImagingControl34 library but it's work with Windows Form and in an IoT app those can't be used.

Did you try something like this or can you give me some advice?

TIS-Stefan, 2019-12-02:
You can use the WindowsFormsHost Control. Alternatively you can use IC Imaging Control dynamically as shown at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/tree/master/c%23/Softwaretrigger-Save-Image. But you must add the System.Windows.Forms to the references, in order to get the live video started.

Stefan

emejoti, 2019-12-03:
The IoT app it's Windows Universal, so I can't use WindowsFormsHost or the alternative. 

Anyway, thank you for the quick answer .

TIS-Stefan, 2019-12-03:
Well, then you may use C++ and the IC Imaging Control C++ Classlibrary? But I think, this is not a real option.

closed 2019-12-03 by TIS-Stefan