TheImagingSource/IC3.5-Imaging-Control-Samples #16: Gain property setting problem with IC Cameras

ljc19800331, 2020-02-01:
Hello, 

I am using the python API for the DFK 33UP1300 camera. When I run the "tis-OpenCV.py" demo code, it shows the error related to the "Camera.SetPropertyValue("Gain","Value",10)" line of code. I am not sure if this is a normal issue with the code or the camera hardware. This is the screen shot: 
[image]
[image]

Thanks.

TIS-Stefan, 2020-02-03:
Hello GuangShen

please check with the 
tcam-ctrl -p 34910523
whether it shows "Gain" property.
Also in the Python sample is an commented call to listproperties, you may enable that and check for Gain. 
Which computer do you use?

Thank you in advance!

Stefan

TIS-Stefan, 2020-02-03:
Hello GuangShen

I just see, its Windows, not Linux.
The Gain range is  144 to  1200 . Therefore you get an error, if you try to set a value of 10.

Stefan

closed 2020-02-03 by TIS-Stefan