TheImagingSource/IC3.5-Imaging-Control-Samples #17: Pycharm searching file for tisgrabber_x64.dll

ljc19800331, 2020-02-19:
Hello, when I run the tisgrabber program in Pycharm I am getting the "[Error 126] The specified module could not be found" error. I have to change the working folder to that folder which contains the "tisgrabber_x64.dll" file. I am wondering if there is a faster way to load the .dll file without changing the working folder? Thanks.

TIS-Stefan, 2020-02-20:
The clean solution is having always a correct working directory being specified.

Well.. The dirty solution is to copy the DLL into one of the directories stored in PATH environment, e.g. c:\windows\system32

Stefan

ljc19800331, 2020-02-26:
Thanks for your help and the issue is caused by the wrong setting working repository in pycharm.

closed 2020-02-26 by ljc19800331