TheImagingSource/IC3.5-Imaging-Control-Samples #18: capture with mingw?

pepeverde9, 2020-04-24:
Hello
I have an optical inspection application written in C++/QT, using gcc on linux and mingw on windows. Now I need to add support for DFK 33UX183 cameras, as we bought some of them to be used for this task, under windows.
But I can't find any documentation about using mingw (or any other C++ besides the one in Visual Studio) with TheImagingSource cameras. IC-Imaging-Control seems to be only about .NET applications, and for tisgrabber I can only find a chm enumerating all the structures and functions, but no example or explanation about how to do something using this library.
As far as I can tell, this camera is not recognized by windows as a webcam to be used with opencv, as another TheImagingSource camera we bought a few years ago was working.

TIS-Stefan, 2020-04-24:
Hello 
For mingw please use the tisgrabber.dll from https://www.theimagingsource.com/support/downloads-for-windows/software-development-kits-sdks/tisgrabberdll/

The make file could be like this:
```
LDFLAGS=-L"$(USERPROFILE)\Documents\The Imaging Source Europe GmbH\TIS Grabber DLL\bin\win32" -ltisgrabber
INC=-I"$(USERPROFILE)\Documents\The Imaging Source Europe GmbH\TIS Grabber DLL\include"

opendevice.exe : main.o 
	gcc main.o $(LDFLAGS) -o opendevice.exe 

main.o :  main.c
	gcc $(INC) -c main.c

```
The main.c like this
```
#include <stdio.h>
#include <conio.h>

#include <tisgrabber.h>

int main()
{
    HGRABBER hGrabber; // The handle of the grabber object.

    IC_InitLibrary(NULL);
    hGrabber = IC_CreateGrabber();
   
    if( hGrabber )
    {
        IC_OpenVideoCaptureDevice(hGrabber,"DMK 23UV024");
        if( IC_IsDevValid(hGrabber))
        {
            IC_StartLive(hGrabber,1);
            printf("Press any key to stop the live video\n" );
            _getch();
            IC_StopLive(hGrabber);
        }
        IC_ReleaseGrabber(&hGrabber);
    }

    return 0;
}
```
Samples can be found in the "ansi.c" code installed into the users directory next to the dll.

pepeverde9, 2020-04-24:
I made a mistake, as the windows 10 Camera application was not recognizing the camera, I supposed that opencv will not work with it. It turns out that it works. By default the images were black and white, but by setting the fourcc property to "YUY2" capturing images just worked.

Thank you for the answer and the short intoductory example. I tested the code, and it works, and maybe use this approach in the future, if I will ever need more than just basic image capturing. But for now I will leave the application as it is, it gets the job done, and there is still work to be done for the core functionality.

Best regards

closed 2020-04-24 by pepeverde9