TheImagingSource/IC3.5-Imaging-Control-Samples #19: Zero value

julienGautier77, 2020-04-28:
Hi 
When I use Tis-openCV-py only a black screen appear. 
The camera is working  with IC capture software
What is happening.  
I use a PC   with windows 10 64 bits with python 3.7.6 32 bit
Do you think it is a 32/63. bit problem
Thanks a lot

TIS-Stefan, 2020-04-28:
You may let me know, which camera you use and whether the sample reports, whether it is valid.
And it is not a 32 or 64 bit issue.

julienGautier77, 2020-04-28:
I 'm using a DMK 42BUCO3 
Perhaps the camera is waiting a trigger an pass the timeout but I check trigger is not set

TIS-Stefan, 2020-04-28:
If you use a trigger, you better try the callback samples from the same repository. They must work...

julienGautier77, 2020-04-28:
I don't use a trigger but it seems that the camera is waiting...

TIS-Stefan, 2020-04-28:
In the devicedialog is properties dialog. Click on it and make sure, that the trigger mode is not enabled.

julienGautier77, 2020-04-28:
The trigger is not set
I add a print(Image) in the tis-OpenCV.py and it is a array of 0
Thanks for your help

TIS-Stefan, 2020-04-28:
The sample uses
`Camera.ShowDeviceSelectionDialog()`
I suppose, the camera is show, you can select it and you can set a frame rate. Use a lower frame rate

From what I see, there must be one Window opened labeled as "ActiveMovie Window" and leater there should be the OpenCV Window. Is that the case?
Remove all the property setting code from the sample, so we strip all, what is not necessary.
If you set a breakpoint at
`Camera.SnapImage()`
And debug step by step, the numpy array in image is null?

Do you use usual Python from Windows Store?

julienGautier77, 2020-04-28:
I just set :
camera=IC.TIS_CAM()
Devices=Camera.GetDevices()
if Camera.IsDevValid()==1 :
Camera.ShowDeviceSelectionDialog() # I set frame rate to 7.35 trigger is not set auto exp is off
camera.StartLive(0)
camera.SnapImage()
image=camera.GetImage()
print (Image)

image=[[000][000][000].....

I use anaconda distribution with spyder (4.1.2) : python 3.7.6 [MSC v.1916 32 bit (intel]
Ipython 7.13.0

TIS-Stefan, 2020-04-28:
Please try
camera.StartLive(1)
And see, what happens.

julienGautier77, 2020-04-28:
an white window is opennig and the windows is still white even without light

TIS-Stefan, 2020-04-28:
Hmm. The Window should be black.... Is it the "ActiveMovie" Window?
Please enable all automatics on the camera. Maybe you use an extremely  long exposure time.

I am sorry for the issues you encounter, but you are the first one having them so I must guess around.

julienGautier77, 2020-04-28:
No problem and thanks for your help again
All is set to auto mode (default)
The activeMovie Windows is still white
I'was able to read this camera before with your software an another computer in my lab (but it'is close..)
Perhaps I have to install  an older version of python (but it will take some time ) because I'm using the latest python version  on my home computer and an older one in my lab (I think)

TIS-Stefan, 2020-04-29:
Hello

It is no Python issue. The ActiveMovie Window must show a live video.
Please start the camera with IC Capture again.
Then use the python sample and remove all stuff, except the opening starting, waiting and stopping the camera. Use the same video format and frame rate, you used in IC Capture.
That must work, because it has no relation to Python at all.

julienGautier77, 2020-04-29:
Hello
I start again with all the same parameter same video format and same frame rate
I just use. the following code 
import tisgrabber as IC
Camera=IC.TIS_CAM()
Devices=Camera.GetDevices()
Camera.ShowDeviceSelectionDialog()
Camera.StartLive(1)
ActiveMovie Window open but it is still white 
I also try with other new cameras,  same model  (I just receive 10 new cameras) but I have the same problem

[image]
[image]

TIS-Stefan, 2020-04-29:
I must say, the content in IC Capture also does not look more interesting..
Do you get real images in IC Capture? The title bar says, you get 23 frames per second. 
Maybe all frames are dropped, because they are transferred incompletely. 
The frame drops may are caused by interrupted image data transfer between USB  controller and computer's memory which are caused by CPU idle states.

In case you use an Intel CPU, you may try Processor Idle State Manager. It tries to prevent the CPU going into C3 state. The program can be downloaded http://www.theimagingsource.com/support/downloads-for-windows/software-tools/procidlestateman/

Stefan

julienGautier77, 2020-04-29:
On IC Capture the camera is working for sure when i remove a little the blocker in front of the camera I can see different image. I install your program : not effect still white windows
On IC capture and on dialog box in python I tried all the format and frame rate : all is working in IC capture program but I have still white windows.

TIS-Stefan, 2020-04-29:
I am lost. IC Capture and tisgrabber use nearly the same DLLs.

Teamviewer? If yes, send contact data through https://www.theimagingsource.com/support/contact/

TIS-Stefan, 2020-04-29:
Looks for Kaspersky Endpoint Security blocking the stream. IC Capture and Camera app work, but the copied Sample Demo App does not as well as the Python program.

Stefan

julienGautier77, 2020-04-29:
As you tell I just switch off kaspersky anti virus and now It is working 
😄😄
All this mess for an anti virus
Thanks a lot for your help

closed 2020-04-29 by julienGautier77