TheImagingSource/IC3.5-Imaging-Control-Samples #21: Access camera frames without ICImagingControl (.netcore)

ahigh2, 2020-07-18:
Hi,

I'd like to be able to access camera frames without explicitly initializing an ICImagingControl in .NET. I have succcesfully created an ASPCore web server to serve camera frames, however it is reliant on this Windows.Forms control.

The ability to access frames without the overhead of the UI component would allow running a "headless" containerized server to serve frames. Is this possible now?

Thanks,
Aaron

TIS-Stefan, 2020-07-20:
Hello Aaron

We are very sorry, but currently this is not possible.

We have plans to decouple the functionality of the WinForms based .NET control from the device interfacing and the frame capture part, but there is no deadline yet. These plans are currently on hold, due to other projects.

If you want to build it yourself, you either can directly wrap the C++ interface or use out C tisgrabber DLL (https://www.theimagingsource.com/support/downloads-for-windows/software-development-kits-sdks/tisgrabberdll/)

Stefan

TIS-Stefan, 2020-07-20:
Hello

I wonder, whether you can get frames or not with your current solution. 

Stefan

ahigh2, 2020-07-20:
Yes, I actually can by just initializing an ICImagingControl outside of an actual Form/Window. My end goal is/was to try to Docker-ize this to a Linux container, which I _can't_ do because of the Windows.Forms dependency. I suspect if I write a C++ wrapper I probably could, so I may explore that avenue.

closed 2020-07-20 by ahigh2