TheImagingSource/IC3.5-Imaging-Control-Samples #22: binning in python

tomasaidukas, 2020-07-22:
Is it possible to enable 2x pixel binning in python? Thanks

TIS-Stefan, 2020-07-22:
Yes, if it is supported by the camera.
I suppose, you use PyhtonNet. Then you simply set the video format as
ic.VideoFormat = "RGB32 (640x480) [Binning 2x]"
Make sure, that the resolution e.g. 640x480 is max the half of the sensor resolution.

tomasaidukas, 2020-07-22:
Thanks Stefan, the cameras I am using support 2x binning, but standard python implementation gives the following video formats once GetVideoFormats() function is called.

TIS-Stefan, 2020-07-22:
The get video formats does not list the binning formats. 
If you want them, you need the VideoFormatDescs object and enumerate the video format descriptions. 

What is "standard python" implementation?
Stefan

tomasaidukas, 2020-07-22:
Oh I see. Just tested and Camera.SetVideoFormat("Y16 (1216x1024) [Binning 2x]") works despite not showing up. By standard I meant standard Python code (not PythonNet). Thanks for the help.

closed 2020-07-22 by tomasaidukas

TIS-Stefan, 2020-07-22:
The code based on tisgrabber.dll supports setting of video formats too. 
But if you use Python 3.7, then the PyhtonNet variant is the better choice, because it is more sophisticated. It can als be used without Qt5, even if my samples all use that. 
And the documentation of IC Imaging Control .NET is far better, than the documenation of the tisgrabber.dll.

Stefan

TIS-Stefan, 2020-07-22:
erld

E-MAIL 

 On Wed, 22nd Jul 2020 at 16:01, TheImagingSource/IC-Imaging-Control-Samples <reply@reply.github.com> wrote: 

Closed #22 [https://github.com/TheImagingSource/IC-Imaging-Control-Samples/issues/22]. 
—

  

 On Wed, 22nd Jul 2020 at 16:01, TheImagingSource/IC-Imaging-Control-Samples <reply@reply.github.com> wrote: 

Oh I see. Just tested and Camera.SetVideoFormat("Y16 (1216x1024) [Binning 2x]") works despite not showing up. By standard I meant standard Python code (not PythonNet). Thanks for the help. 

  
 Was this message helpful?   Yes   [https://theimagingsource.deskpro.com/ticket-rate/VBKW-6126-NMYP/6W82G4DN8MZWTAN/194265?setrating=1]   It was OK   [https://theimagingsource.deskpro.com/ticket-rate/VBKW-6126-NMYP/6W82G4DN8MZWTAN/194265?setrating=0]   No   [https://theimagingsource.deskpro.com/ticket-rate/VBKW-6126-NMYP/6W82G4DN8MZWTAN/194265?setrating=-1] 

View and manage this ticket online: https://theimagingsource.deskpro.com/tickets/39123 

  The Imaging Source [https://www.theimagingsource.com] · https://theimagingsource.deskpro.com/ 

 (#CFWT6W82G4DN8MZWTAN)

tomasaidukas, 2020-07-22:
Python at the moment seems to do the job. Can I expect performance enhancement from PythonNet or just better documentation etc? If yes, then I will try to use it instead. Thanks.

TIS-Stefan, 2020-07-22:
The .NET component is more sophisticated and has one image copy less, if the FrameNotificationSink is used. 
Also by using the .NET component, the latest IC Imaging Control is in use. And you have the all functions of the SDK.
The tisgrabber.dll is somewhat older. It does not implement all functions available by IC Imaging Control SDK and it also needs always a Python wrapper, that must adapt to the changed in tisgrabber.dll. Which means double work for me. 
Therefore, I consider end  of development of  tisgrabber.dll for Python.

However, the PythonNet is currently limited to Python 3.7. I do not know, when they make a 3.8 version. 

Stefan

tomasaidukas, 2020-07-23:
Hi Stefan,

Thank you. Currently I just turned on the cameras in trigger mode and ran them till the required number of frames can be captured using a callback function. Are there any reference on how this could be done with PythonNet. I am a bit unsure which object need to be accessed for that from the sink/listener/ic classes. If you could point me in the right direction that would be great.

Best wishes
Tomas

TIS-Stefan, 2020-07-23:
Hello Tomas

you would use the FrameQeueSink for this task:
https://www.theimagingsource.com/support/documentation/ic-imaging-control-net/FrameQueueSink.htm

You can create in the FrameQeueListener a frame queue with the wanted number of images. 

Stefan