TheImagingSource/IC3.5-Imaging-Control-Samples #23: FrameQueueBufferImpl.Dispose leads to stack overflow.

hglee, 2020-09-22:
Hi, I'm trying to use FrameSnapSink introduced in v3.5.

When I dispose result of SnapSingle method, it leads to stack overflow.

Sample code:

```csharp
using (var buffer = this.Sink.SnapSingle(TimeSpan.FromMilliseconds(timeoutMillisec)))
{
    return buffer.CreateBitmapCopy();
}
```

And decompiler shows self call in Dispose method.

[image]

TIS-Stefan, 2020-09-22:
You are right, this issue has to be fixed on our side.

We will try to fix this in the near future.

Thanks for reporting this issue.

hglee, 2020-11-16:
Works without problem in new release, v3.5.6.

We can close this.

Thanks.

closed 2020-11-16 by hglee