TheImagingSource/IC3.5-Imaging-Control-Samples #24: Python not responging after clicking the Live window

JesusDelHoyo, 2021-01-27:
Hello.

I have a problem. After opening the window using the StartLive method, if the window is made the active one (for example, by clicking on it), Python stops working:
[image]
Can anyone help me to solve it?

Thanks.

TIS-Stefan, 2021-02-19:
Sorry for the delay of my answer.
From the image alone I can not guess anything.
Which camera model, video format and frame rate to you use?
Where does Python hang?
Which Python version do you use?

Stefan

JesusDelHoyo, 2021-02-22:
The camera is: DMx 72BUC02. I use it with the settings it is initialized (max. resolution and framerate).
I tried using Python 2.7 and 3.8-3.9. I had the same problem with both of them.
Python hangs as soon as the active movie window becomes the active window (if I click the icon or I close other windows).

Jesus.

TIS-Stefan, 2021-02-22:
Hello Jesus

Thank you. Can you please install IC Capture and see, whether you can use the camera with it? IC Capture can be downloaded from https://www.theimagingsource.com/support/downloads-for-windows/end-user-software/iccapture/.

Which Python sample did you use? (It does not look for the PythonNET sample, which I would prefer.)

Stefan

JesusDelHoyo, 2021-02-22:
Hello Stefan

I use IC Capture normally without problems.

I'm not sure if I understand what do you mean as a Python sample. Mainly, I use the module tisgrabber.py. 

Thank you for your help.
Jesus.

TIS-Stefan, 2021-02-22:
Hello Jesus

Can you please show me the smallest python script you use, that shows your issue?
Does the live video update in the "ActiveMovie" window?

Stefan

JesusDelHoyo, 2021-02-22:
A shortened version of the script is this:

```
import ImagingSource.tisgrabber as IC
cam = IC.TIS_CAM()
cam.open("DMx 72BUC02 14210296")
cam.StartLive(1)
```

The video in the ActiveMovvie window updates correctly until the window becomes the active one and python crashes.

Jesus.

TIS-Stefan, 2021-02-22:
Hi Jesus

No call to stoplive and exitlibrary?

Can you add this please?

Stefan

JesusDelHoyo, 2021-03-01:
Hi Stefan.

I use a Jupyter Notebook to control the camera. The part of the code to take the images (SnapImage) and stop the camera (StopLive) are afterwards. 

Jesus.

TIS-Stefan, 2021-03-01:
Hi Jesus

can you please try with "normal" Python? I am not familiar with Jupyter Notebook and do not know, which side effects that has.

Stefan

closed 2021-07-28 by TIS-Stefan