TheImagingSource/IC3.5-Imaging-Control-Samples #25: OpenCV example code crashes

zurxi, 2021-03-20:
When I try to run the example code, line 79 in tis-OpenCV.py crashes.

`Camera.SetPropertyValue("Gain","Value",10)`

The output is:

`Traceback (most recent call last):
  File "D:/rodma/Documents/School/EVD/IC-Imaging-Control-Samples/Python/Open Camera, Grab Image to OpenCV/tis-OpenCV.py", line 79, in <module>
    Camera.SetPropertyValue("Gain","Value",10)
  File "D:\rodma\Documents\School\EVD\IC-Imaging-Control-Samples\Python\Open Camera, Grab Image to OpenCV\tisgrabber.py", line 659, in SetPropertyValue
    error = TIS_GrabberDLL.SetPropertyValue(self._handle,
OSError: [WinError -529697949] Windows Error 0xe06d7363`

If this is commented out, everything else still works. Is there an issue with my machine? The camera works properly in IC Capture.

TIS-Stefan, 2021-03-22:
Which camera model do you use?
Mayby Gain is not a long, but a float value, so you may pass 10.0?

zurxi, 2021-03-29:
I'm using the camera model DFKZ12G445. I tried testing with a float, and I got a wrong type error, so I don't think that fixes it. The workaround I'm using right now is the set the Gain to manual in the code, then go into IC Capture, set the gain manually, then run the code. The gain value is carried over that way, but the SetPropertyValue function doesn't seem to work with gain.

TIS-Stefan, 2021-03-29:
First of all, I am sorry, I thought, we are in Linux, but we are in Windows here.

The value 10 is out of Gain Range for integer values:
```
Gain
    Gain Range        Range                  200     1023      200
    Gain Range        Absolute Value           0    29,46        0
```
So minimum is 200. 
Gain has two interfaces: The "Range" interface with integer values and the Absolute Values with float values in dB.

Please set your gain to 200 and try again.

Stefan

zurxi, 2021-04-16:
Hi, that worked! Thank you so much.

closed 2021-04-19 by TIS-Stefan