TheImagingSource/IC3.5-Imaging-Control-Samples #27: Camera setting changing after first image capture

temp369, 2021-04-08:
While trying to capture the image using python code I am able to get the correct image as per the settings in IC capture software for the first time. But for the second time if I try the same code I am getting some unclear image. If I open and close IC capture software I will be getting the proper image.  How to fix this? I also tried the sample code and got the same results.

`    
 
    Camera.open("DFK yyyy xxxxx")

    Camera.SetContinuousMode(1)

    Camera.StartLive(0) 

    Camera.SnapImage()

    Camera.StopLive() 

    Camera.SaveImage(Filename,"JPEG")`

TIS-Tim, 2021-04-08:
Can you explain in more detail what "unclear image" means?

temp369, 2021-04-09:
Looks like the camera settings changed.  I have adjusted the settings in IC capture software for better vision and on running the python script I get a proper image for the first time but for the second time, I a getting an image that looks likes the image settings has changed. If I open and close the software I am getting the proper image.

TIS-Tim, 2021-04-09:
You should implement proper settings management in your program instead of hoping that the device will keep its state between executions.
The simplest was to do that:
- Call ShowPropertyDialog. This will display the same dialog that you use in IC Capture to configure device settings
- Call SaveDeviceStateToFile( <filename> ). This Will save the current camera settings into a file
- On startup, call LoadDeviceStateFromFile( <filename> ). This will open the previously opened device and restore all camera settings

temp369, 2021-04-09:
I tried without ShowPropertyDialog() and it's doesn't made a difference.  When I tried with ShowPropertyDialog I am getting the error "exception: access violation reading 0x0000000000000001" from tisgrabber.py" after giving "ok" in dialog window.

NB = Mine is a dark environment with a display

TIS-Stefan, 2021-04-12:
Hello

Please let us know, which camera model you use. 
The access violation is unexpected. Where in your code did you use that? I suppose, it is after Camera.open() call, which is, of course, necessary

Stefan

temp369, 2021-04-16:
Model No: DFK Z12GP031

I called it after Camera.open().

I tried to save settings without ShowPropertyDialog() and loaded in the next run. It's doesn't made a difference.

TIS-Stefan, 2021-04-16:
Hello
Please use the sequence below:

Camera.open("DFK yyyy xxxxx")
Camera.SetContinuousMode(1)
Camera.StartLive(0) 
Camera.SnapImage()
Camera.SaveImage(Filename,"JPEG")`
Camera.StopLive() 

If the access violation still occurs, please let me know, in which exact line of code it happens.
Thank you in advance!

Stefan

erte0011, 2021-04-23:
Hello,

I have the same access violation issue. The following works fine (no property dialog);
```
cam.open("DFK yyyy xxxxx") 
cam.SetContinuousMode(1)
cam.StartLive(0)
time.sleep(1.0)
cam.SnapImage()
cam.StopLive()
cam.SaveImage(Filename, "JPEG")
```

However, adding ShowPropertyDialog() results in the access violation;
```
cam.open("DFK yyyy xxxxx")
cam.ShowPropertyDialog()
cam.SetContinuousMode(1)
cam.StartLive(0)
time.sleep(1.0)
cam.SnapImage()
cam.StopLive()
cam.SaveImage(Filename, "JPEG")
```

"...\tisgrabber.py", line 470, in SetContinuousMode
    return TIS_GrabberDLL.SetContinuousMode(self._handle, Mode)
OSError: exception: access violation reading 0x0000000000000001

Note, if you instead access the property dialog from ShowDeviceSelectionDialog it works fine;
```
cam = tisgrabber.TIS_CAM()
cam.open("DFK yyyy xxxxx")
# cam.ShowPropertyDialog()
cam.ShowDeviceSelectionDialog()  # Clicking properties button in this dlg works fine
cam.SetContinuousMode(1)
cam.StartLive(0)
time.sleep(1.0)
cam.SnapImage()
cam.StopLive()
cam.SaveImage(Filename, "JPEG")
```

I can provide more info if you need it.

Thank you,
-Robert

TIS-Stefan, 2021-04-23:
Hello

Can you please add a source code file, so I can try to debug this on next Monday (it is 5:50 pm now, so I wont start this today, Sorry.)

Stefan

erte0011, 2021-04-23:
Sure, please see attached zip containing minimal_example.py
Thank you!

[minimal_example.zip](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/6366736/minimal_example.zip)

erte0011, 2021-04-23:
Hello, I think I have a fix; looking at tisgrabber.h "IC_ShowPropertyDialog", it appears the return type is int (success, error, etc.), not hgrabber handle. Updating tisgrabber.py as follows makes it work;

Modify Dll interface like this;
```
ShowPropertyDialog = __tisgrabber.IC_ShowPropertyDialog
# ShowPropertyDialog.restype = GrabberHandlePtr
ShowPropertyDialog.restype = C.c_int
ShowPropertyDialog.argtypes = (GrabberHandlePtr,)
```

Modify associated method like this;
```
def ShowPropertyDialog(self):
    # self._handle = TIS_GrabberDLL.ShowPropertyDialog(self._handle)
    return TIS_GrabberDLL.ShowPropertyDialog(self._handle)
```

I tested and it is working on my setup. I can do a pull request if it makes more sense.

Thanks,
-Robert

TIS-Stefan, 2021-04-26:
Dear Robert

Thank you for your email.
Lets do the (for me) obvious issue first:
The line

cam.SetContinuousMode(1)must be changed to

cam.SetContinuousMode(0)

The continuous mode = 1 saves each image into the internal ringbuffer, therefore snapImage() will fail.

The not obvious error is in tisgrabber.py, line 741
The original is

return TIS_GrabberDLL.SaveImage(self._handle, self.s(FileName), IC.ImageFileTypes[self.s(FileType)],Quality)

The first issue is, that "IC" is not defined. I do not know, why that worked previously, however, if "IC" is removed, we make a small step forward:

return TIS_GrabberDLL.SaveImage(self._handle, self.s(FileName), ImageFileTypes[self.s(FileType)],Quality)"ImageFileTypes" is defined on top of "tisgrabber.py".

Now we receive a "Exception has occurred: KeyError", which lets us know, that "JPEG" is not a key of 

ImageFileTypes = {'BMP':0, 'JPEG':1}That is quite confusing, but the reason is, that the input string is converted to utf-8 which is not a string, but a byte array. Visual Studion code debugging shows it as "b'JPEG'". Therefore, the simple solution is removing the self.s(FileType) call:
return TIS_GrabberDLL.SaveImage(self._handle, self.s(FileName), ImageFileTypes[FileType],Quality)

That works. The reason of using self.s() call was compatibility between Python2.7 and Python 3.x. 

So you may enhance your tisgrabber.py. I do not add that to the repository, because I am not clear about the versioning of Python as well about, why I get a byte array and not a string for use as key in ImageFileTypes. I have to admit, I am not a Python professional.

Yours sincerely

Stefan Geissler
--------------------------------------------------------------------
The Imaging Source Europe GmbH
Ueberseetor 18
D-28217 Bremen
Germany
Tel   +49/421/33591-0
Fax   +49/421/33591-80
Registergericht: Amtsgericht Bremen HRB 11 927
Geschäftsführer: Rolf Bollhorst 
--------------------------------------------------------------------

E-MAIL 

  
 Was this message helpful?   Yes   [https://theimagingsource.deskpro.com/ticket-rate/JHEM-3542-XVPT/TP6GDN9P45KQ3Y2/290793?setrating=1]   It was OK   [https://theimagingsource.deskpro.com/ticket-rate/JHEM-3542-XVPT/TP6GDN9P45KQ3Y2/290793?setrating=0]   No   [https://theimagingsource.deskpro.com/ticket-rate/JHEM-3542-XVPT/TP6GDN9P45KQ3Y2/290793?setrating=-1] 

View and manage this ticket online: https://theimagingsource.deskpro.com/tickets/56994 

  The Imaging Source [https://www.theimagingsource.com] · https://theimagingsource.deskpro.com/ 

 (#DGICTP6GDN9P45KQ3Y2)

TIS-Stefan, 2021-04-26:
Hello Robert.

You are absolutely correct about the the showPropertydialog call in tisgrabber.py. This is a obvious, stupid copy&paste error. I am very sorry.

However, can you help me regarding what I wrote above about using the ImageFileTypes array?

Thank you in advance!

Stefan

erte0011, 2021-04-29:
As you mention, the 's' method is there for compatibility. In Python 2, strings are already bytes and can be used as-is with ctypes 'c_char_p', whereas in Python 3 strings need to be encoded as a byte string to work, hence the 'strin.encode("utf-8")' line in 's'. So the 's' method is important and used a lot for interfacing with tisgrabber.dll.

In the specific case of the 'ImageFileTypes' dict lookup in 'SaveImage'; as you already said, in Python 3 the key will end up being a byte string after going through the 's' method, and it won't work because the dict is expecting a regular string. In Python 2.7 since the 's' method will not modify the key, then the dict lookup will work. I don't have Python 2.7 installed so I can't test this, but I assume that's why SaveImage was working when this was first implemented/tested. So in summary my opinion is removing the 's' from the 'ImageFileTypes' lookup is the smallest/cleanest solution, with the least impact, that works with both Python 2.7 and 3.x.

I hope this helps answer your question. Also I don't use SaveImage in my specific application, I was just using it as part of the minimal example, so it's not urgent for me at least.

Thanks,
-Robert

TIS-Stefan, 2021-04-29:
Hi Robert

thank you for your explanation and the recommendation. I will update the tisgrabber.py as soon as possible.

Stefan

temp369, 2021-05-03:
I solved the issue. Later I found that WDR setting is causing the un-stability of the images. When I tried to calibrate the image without WDR, I am getting stable image.

Thank you for the time 👍

closed 2021-05-03 by temp369