TheImagingSource/IC3.5-Imaging-Control-Samples #28: Explicitly set the gain using the SetPropertyValue value.

s-koich, 2021-04-13:
I am using tisgrabber.py on a DMK 33UP1300 with Windows.
I am trying to turn off auto gain.
So far I have used calls like Camera.SetPropertyValue('Gain', 'Value', 150).
However, the default value of Gain, 144, is set.
The return code is also -7 and will fail.
Can you tell me why?
For example, it would look like this

import time
import tisgrabber as IC

# Set up camera
Camera = IC.TIS_CAM()

Camera.open('DMK 33UP1300 22020377') # insert real serial number...
Camera.SetVideoFormat('Y800 (1280x1024)')
Camera.SetFrameRate(8)
Camera.StartLive(0)

print("Set GainVideoAuto Returned:" , Camera.SetPropertySwitch('Gain', 'Auto', 0))

# Set Gain value
print('Set Gain Returned: ' + str(Camera.SetPropertyValue('Gain', 'Value', 150)))

# Get Gain value
print('Gain: ' + str(Camera.GetPropertyValue('Gain', 'Value')))

# wait a bit
time.sleep(0.5)

# query again
print('Gain: ' + str(Camera.GetPropertyValue('Gain', 'Value')))

Camera.StopLive()

TIS-Stefan, 2021-04-13:
Hello

First of all, thank you very much for the sample code, that saved a lot of guessing and work for me!

I tried the code and it works as expected, I always get the gain value back which I set before.

Please check, whether you installed the correct driver from https://www.theimagingsource.com/support/downloads-for-windows/device-drivers/icwdmuvccamtis33u/

s-koich, 2021-04-13:
Thanks for the response.

I am glad that the code is correct.
I tried to test with the same code, but it set failed.
==
Execution result
Set GainVideoAuto Returned: 1
Set Gain Returned: -7
Gain: 144
Gain: 144
==

As you can see in the figure below, I think the driver is installed correctly.

[image]

As you can see in the figure below, I think the driver is installed correctly.

However, when I connect the camera, it says that the parameters are different.

Is this normal behavior?

TIS-Stefan, 2021-04-14:
Hello

I agree, the driver for USB 33 cameras from the link I mentioned, seems to be installed. You could make sure, that it is version 5.0.1.

The error -7 indicates, that the value to be set is out of the property range from min to max. The internal tisgrabber C++ source code is:
```
	/////////////////////////////////////////////////////////////////////////////////
	// Set the current value of a given property and element combination. 
	// Range interface is used
	// Property and Element names are not case sensitive
	int setPropertyValue(DShowLib::Grabber &Grabber, char* Property, char *Element, int Value)
	{
		int iSuccess = IC_SUCCESS;
		tIVCDRangePropertyPtr pRange = NULL;
		
		iSuccess = GetRangePropertyPtr(Grabber, Property, Element, pRange);
		if (iSuccess == IC_SUCCESS)
		{
			if (pRange != NULL)
			{
				if (pRange->getRangeMin() >= (long)Value && (long)Value <= pRange->getRangeMax())
					pRange->setValue((long)Value);
				else
					iSuccess = IC_INDEX_OUT_OF_RANGE;
			}
		}
		return iSuccess;
	}
```
IC_INDEX_OUT_OF_RANGE is defined as -7.
That  means, that the Gain Value property was found, but the value we tried to set was out of range. But the value of 150 is clearly in the range of the Gain Value property. 
What happens, if you set a value of 144?

Could you try the Gain Absolute and set a value of e.g. 2.0?

` Camera.SetPropertyAbsoluteValue("Gain","Value",2.5)`

I am a little bit puzzled.

Stefan

s-koich, 2021-04-14:
Thank you for your response.

I also re-installed the driver, so it is definitely version 5.0.1.
[image]

>What happens, if you set a value of 144?
OK  Successful

>Set GainVideoAuto Returned: 1Could you try the Gain Absolute and set a value of e.g. 2.0?

>Could you try the Gain Absolute and set a value of e.g. 2.0?
Set Absolute Gain Returned: -7

>Camera.SetPropertyAbsoluteValue("Gain","Value",2.5)
Set Absolute Gain Returned: -7

>I am a little bit puzzled.
me too.

==
import time
import tisgrabber as IC
import cv2

# Set up camera
Camera = IC.TIS_CAM()

Camera.open('DMK 33UP1300 22020377') # insert real serial number...
Camera.SetVideoFormat('Y800 (1280x1024)')
Camera.SetFrameRate(8)
Camera.StartLive(0)

GainAblolutevalue = [0]

print("Set GainVideoAuto Returned:" , Camera.SetPropertySwitch('Gain', 'Auto', 0))

# Set Gain value
print('Set Gain Returned: ' + str(Camera.SetPropertyValue('Gain', 'Value', 144)))

# Set Absolute Gain value
print('Set Absolute Gain Returned: ' + str(Camera.SetPropertyAbsoluteValue("Gain", "Value",2.0)))

# Get Gain value
print('Gain: ' + str(Camera.GetPropertyValue('Gain', 'Value')))

# Get Gain value

ret = Camera.GetPropertyAbsoluteValue('Gain', 'Value', GainAblolutevalue)
print('Absolute Gain : ' + str(GainAblolutevalue[0]) + ',Return Code : ' + str(ret) )

# wait a bit
time.sleep(0.5)

# query again
print('Gain: ' + str(Camera.GetPropertyValue('Gain', 'Value')))

Camera.StopLive()

=
Set GainVideoAuto Returned: 1
Set Gain Returned: 1
Set Absolute Gain Returned: -7
Gain: 144
Absolute Gain : 0.0,Return Code : 1
Gain: 144

TIS-Stefan, 2021-04-14:
Hello 

A colleage just showed me, that the code can not work at all:
Wrong:
`
if (pRange->getRangeMin() >= (long)Value && (long)Value <= pRange->getRangeMax())
`
Must be
`
if (pRange->getRangeMin() <= (long)Value && (long)Value <= pRange->getRangeMax())
`

I will correct this during this day.

I am sorry for the error. I do not know, when this entered the code,

Stefan

TIS-Stefan, 2021-04-14:
Inbetween you can download the tisgrabber DLL from https://www.theimagingsource.com/support/downloads-for-windows/software-development-kits-sdks/tisgrabberdll/ and overwrite the DLL files in your Python project directory

TIS-Stefan, 2021-04-15:
Good Morning.

Did you had a chance to follow my latest advice? If yes, what was the result?

Thank you in advance!

Stefan

TIS-Stefan, 2021-04-15:
Today I downloaded the repository and checked the tisgrabber.dll and tisgrabber_x64.dll versions and used these version on my computer. The gain value could be set and read perfectly.

The version of the DLLs is 3.4.0.4 in the repository, which are the expected ones. Please make sure, you use the same versions from this sample repository.

I worked on a newer version, which indeed has an error, as you can see in the source code I pasted. But this newer version was never released.

Can you please check versions?

Thank you in advance!

s-koich, 2021-04-15:
Sorry for the delay. Thanks Stefan.

The DLL version I was using was 3.4.0.5.
The latest version from a Japanese trading company is here.
I was told by a Japanese trading company that the latest version was this one, so I used it.

I'll let the trading company know.

I will let the trading company know, but I would also like to ask Stefan to stop using it.

The DLL version is 3.4.0.4, and it works fine.
Mission complete.

I will post the results as well.

import time
import tisgrabber as IC
import cv2

# Set up camera
Camera = IC.TIS_CAM()

Camera.open('DMK 33UP1300 22020377') # insert real serial number...
Camera.SetVideoFormat('Y800 (1280x1024)')
Camera.SetFrameRate(8)
Camera.StartLive(0)

GainAblolutevalue = [0]

print("Set GainVideoAuto Returned:" , Camera.SetPropertySwitch('Gain', 'Auto', 0))

# Set Gain value
print('Set Gain Returned: ' + str(Camera.SetPropertyValue('Gain', 'Value', 150)))

# Set Absolute Gain value
#print('Set Absolute Gain Returned: ' + str(Camera.SetPropertyAbsoluteValue("Gain", "Value",2.0)))

# Get Gain value
print('Gain: ' + str(Camera.GetPropertyValue('Gain', 'Value')))

# Get Gain value

ret = Camera.GetPropertyAbsoluteValue('Gain', 'Value', GainAblolutevalue)
print('Absolute Gain : ' + str(GainAblolutevalue[0]) + ',Return Code : ' + str(ret) )

# wait a bit
time.sleep(0.5)

# query again
print('Gain: ' + str(Camera.GetPropertyValue('Gain', 'Value')))

Camera.StopLive()

===
Set GainVideoAuto Returned: 1
Set Gain Returned: 1
Gain: 150
Absolute Gain : 0.17000000178813934,Return Code : 1
Gain: 150

TIS-Stefan, 2021-04-16:
Hello 

Yes, it can happen, that the well known company in Japan got a beta version of the DLLs.
(I already fixed that issue in the source code.)

Thank you very much for you patience and help finding an error.

Stefan

closed 2021-04-16 by TIS-Stefan