TheImagingSource/IC3.5-Imaging-Control-Samples #30: Cpp OpenCV windows examples

aditya369007, 2021-06-16:
Hello there,

I can see that there are OpenCV examples on the Linux SDK branch, but I was not able to find any over on this branch.
Can you please guide me on how I can interface this SDK to get images in OpenCV?

Thanks in advance

TIS-Stefan, 2021-06-16:
Hello

If you have the image buffer you use

```
void DoImageProcessing(smart_ptr<MemBuffer> pBuffer)
{
	FrameTypeInfo FTI = pBuffer->getFrameType();

	cv::Mat Matrix(FTI.dim.cy, FTI.dim.cx, CV_8UC3, pBuffer->getPtr());
}
```

That is all.

Stefan

closed 2021-06-16 by TIS-Stefan

TIS-Stefan, 2021-06-16:
Hello

the previous answer was somewhat short. The following would be the IC 3.5 style.
First of all, we assume, the pixel format in the sink has been set to RGB24. Therefore, we will use CV_8UC3 in the cv::Mat creation.
We also need the dimension of the frame and the pointer to the image data. Please refer to 

https://www.theimagingsource.com/support/documentation/ic-imaging-control-cpp/FrameTypeInfo.htm

and

https://www.theimagingsource.com/support/documentation/ic-imaging-control-cpp/meth_descIFrame_getPtr.htm

Then we query these data, create the cv::Mat and pass the image data pointer to it:

```
void CListener::DoImageProcessing(iFrame Frame)
{
	FrameTypeInfo FTI = Frame.getFrameType();
	cv::Mat Matrix(FTI.dim.cy, FTI.dim.cx, CV_8UC3, Frame.getPtr());
}
```

In case you create the cv::Mat without the pointer, but with image data memory allocated, you can use memcpy. This is somewhat slower, but you can be sure, the image data is not overwritten by frames to come. 

I hope, this helps.

Stefan

aditya369007, 2021-06-16:
Thank you I will try it out and reach out to you if I face any issues.