TheImagingSource/IC3.5-Imaging-Control-Samples #31: TIS_GrabberDLL.GetImagePtr() can’t return ImagePtr

yfr-yoyo, 2021-06-24:
Hello.
I've got the problem when running the tis-OpenCV.py. It gave me an error about c and I can't fix it. 
OSError: exception: access violation reading 0x0000000000000000
I got this error when trying to get the image, using image=Camera.GetImage()
Thanks a lot if anyone could help.

TIS-Stefan, 2021-06-24:
Hello

Please let me know, which camera model, video format and frame rate you use. Also: Did you get images at all? Does IC Capture (https://www.theimagingsource.com/support/downloads-for-windows/end-user-software/iccapture/) show images from the camera?

Stefan

yfr-yoyo, 2021-06-24:
The camera is: 33UX174. Video Format, frame rate is GBRG(640x480 VGA) and 312.0027. I'didn't get any image. The window has opened and then quickly closed. I can use IC Capture2.5 to show video and also capture images.  Are there any

yfr-yoyo, 2021-06-24:
The camera is: 33UX174. Video Format, frame rate is GBRG(640x480 VGA) and 312.0027. I'didn't get any image. The window has opened and then quickly closed. I can use IC Capture2.5 to show video and also capture images.

TIS-Stefan, 2021-06-24:
Thank your very much for the answer, that is helpful. Please install the driver from https://www.theimagingsource.com/support/downloads-for-windows/device-drivers/icwdmuvccamtis33u/
That should solve your issue.

Stefan

yfr-yoyo, 2021-06-24:
I have no idea wheather you mean install the latest driver of version 5.0.1.1608. Infact, I have tried remove it from my win10 and install it again, unfortunately it doesn't work.

TIS-Stefan, 2021-06-24:
Yes, I mean, you must install the driver. 
If the video format is "rggb" instead of "RGB24" or "Y800", etc, then the driver is not installed.
Please check in Windows Device Manager, whether the Microsoft or the "The Imaging Source" driver is installed. You may need to force the Device Manager to use our driver. You do that by completely manual driver installation in Device Manager.

Stefan

yfr-yoyo, 2021-06-24:
I really appreciate it！After I restarted my computer, the driver works well now. And by the way, why the ActiveMovie Window and Window both look  blank? There's no image.

TIS-Stefan, 2021-06-24:
The "Active Movie" Window should not be blank. Did you call startlive(0)? I mean, without showing live video?

yfr-yoyo, 2021-06-24:
No, it's   Camera.StartLive(1)

TIS-Stefan, 2021-06-24:
Well, then an image is expected.
Do you get something in the cv::mat? 
Please check, whether the trigger mode is disabled. You can do that with IC Capture.

Stefan

yfr-yoyo, 2021-06-24:
I've found a tick for trigger under other option. After I tick it, I get a white window.   The window is black before.

TIS-Stefan, 2021-06-24:
White means, a lot of light, but data. Very well.

yfr-yoyo, 2021-06-24:
I reset the apeture and it solved! Thank you very much!It's really kind of you to offer me so much help.

TIS-Stefan, 2021-06-24:
Thank you very much for your compliments!

Stefan

closed 2021-06-24 by TIS-Stefan