TheImagingSource/IC3.5-Imaging-Control-Samples #32: The ground truth of DeviceTimeStamp

takuro1026, 2021-06-24:
Hello TIS team,

I have a question regarding DeviceTimeStamp which I got from FrameMetadata::DeviceTimeStamp.
The current project I'm working on, I need to sync 2 videos which recorded by 2 different camera at same time. In order to make this work, I need to check every frame's timestamp and align it, but I also need to know where the timestamp came from?
Is it from device itself? If so, is it possible that two different devices have different time? How can I set the time for a specific device?
Therefore, I can make sure that the timestamp I got from FrameMetadata can be align directly across different devices.

Hope you have a great day! Thank you.

Best,

W.S.

TIS-Stefan, 2021-06-24:
Hello W.S.

If the used device supports "Device Time Stamps", then the time stamp comes, as the name suggests, from the device. Not from the driver. 
The time stamp is generated by the clock of the device and is started, when the device is powered on. It is not reset, when the stream is started, as the Device Frame Number is. Therefore, using the device time stamps of to cameras needs to know the time difference between the cameras, so you can normalize them. 

Which device do you use?

Stefan

takuro1026, 2021-06-24:
Hey Stefan,

The device model I'm using is DFK 33GX287. Based on what you said, it seems unless I can powered both camera at same time; otherwise there is no way to sync it. Another question will be, is there any way to get the current device time? I did check the docs, but in Device class, it only has get serial number method. Thanks.

Best,

W.S.

TIS-Stefan, 2021-06-24:
Hello W.S.

There is no function for getting the device time. 
You can do following:
Set both cameras in trigger mode
Start both cameras.
set a software trigger to both cameras.
Then you get two images.
use their device frame times and get the differences.

Alternatively use the DeviceFrameNumber. From my point of view it is much easier to use. I use that too in the IC Express program.

Stefan

takuro1026, 2021-06-24:
Hi Stefan,

Thank you so much for answering my question.
I'm currently using the similar approach like you provided to get the images; however, I'm using multiprocessing to control each camera, so it does not guarantee that the software trigger will be push at the same time. 
Since my project is using to detect athlete's pose and via two camera to rebuild the scene which means I use 2 different views to reconstruct the 3 dimension coordinate, so the accuracy has high standard and that's why I'm seeking a more reliable way to align the frames.

I will think about it and thank you again for providing me your solution.

All the best,

W.S.

TIS-Stefan, 2021-06-24:
Hello W.S.

If I make a program like this, I use trigger mode and either hardware trigger or software trigger for triggering the camera. 
I would use the DeviceFrameNumber for finding the pairs of images. 

I suppose, you have a main program, that controls both cameras. That could call the software triggers

```
void Triggerthread( void* SoftTrigger )
{
	tIVCDButtonPropertyPtr *st = (tIVCDButtonPropertyPtr*)(SoftTrigger);
	(*st)->push();
	
	_endthread();
}

void CCameraCtrl::Trigger()
{
	if(m_cGrabber.isDevValid() && m_cGrabber.isLive())
	{
		if( m_pSoftwareTrigger != NULL)
		{
			_beginthread( Triggerthread, 0, (void *)&m_pSoftwareTrigger  );
		}
	}
}

```

In the main program I have
```
CCameraCtrl cams[2];
...
cams[0].Trigger();
cams[1].Trigger();
```
Using an own thread for software trigger push is the fastest method sending two triggers in sequence. The push() functions sends the command immediately to the camera and the camera starts exposing, but it can take 4 to 7ms until the push() function returns. This time can be spared by using threads for the push() call.

Stefan

closed 2021-07-28 by TIS-Stefan