TheImagingSource/IC3.5-Imaging-Control-Samples #33: sometimes get NULL point from tisgrabber.GetImage.GetImagePtr()

adamtsai66, 2021-07-06:
Hello,
We use DMK 33UX226 camera, and use python code to init camera and get image. We init camera sample code in below. 
After init camera, we call Getimage() to get the image data, but sometimes we get NULL point when we call Getimage.getimagePtr().
I would appreciate it if you could give some help to us.

    #-------------------------------------------
        def Init_camera():
            lWidth = C.c_long()
            lHeight = C.c_long()
            iBitsPerPixel = C.c_int()
            COLORFORMAT = C.c_int()
            #--------------------------
            # Create the camera object.
            #--------------------------
            Camera = IC.TIS_CAM()
        
            #-------------------------------------------
            # List availabe devices as uniqe names. This is a combination of camera name and serial number
            #-------------------------------------------
            Devices = Camera.GetDevices()
            for i in range(len( Devices )):
                print( str(i) + " : " + str(Devices[i]))
        
            #-------------------------------------------
            # Open a device with hard coded unique name:
            #-------------------------------------------
            Camera.open(bytes.decode(Devices[i]))
            if Camera.IsDevValid() == 1:
                system_log.error("device be found")
        
                #--------------------------------
                # Load default of Camera property
                #--------------------------------
                #Camera.SaveDeviceStateToFile('camera_property.txt')
                with open('camera_property_read.txt','rt',encoding="utf-8") as myfile:
                    data = myfile.read()
                with open('camera_property_write.txt','wt',encoding="utf-8") as myfile:
                    print(data.replace('DMK 33UX226 31914288', bytes.decode(Devices[i])))
                    myfile.write(data.replace('DMK 33UX226 31914288', bytes.decode(Devices[i])))
                Camera.LoadDeviceStateFromFile('camera_property_write.txt')
        
                #-------------------
                # Set a video format
                #-------------------
                Camera.SetVideoFormat("Y800 (4000x3000)")
        
                #-----------------------------------------
                # Set a frame rate of 30 frames per second
                #-----------------------------------------
                Camera.SetFrameRate(5.0)
        
                #--------------------------
                # set iBytesPerPixel=8  (1)
                #--------------------------
                Camera.SetFormat(IC.SinkFormats.Y800)
        
                ExposureAuto = [1]
        
                Camera.GetPropertySwitch("Exposure","Auto",ExposureAuto)
                print("Exposure auto : ", ExposureAuto[0])
                #------------------------------------------------------------------------------------------
                # In order to set a fixed exposure time, the Exposure Automatic must be disabled first.
                # Using the IC Imaging Control VCD Property Inspector, we know, the item is "Exposure", the
                # element is "Auto" and the interface is "Switch". Therefore we use for disabling:
                #------------------------------------------------------------------------------------------
                Camera.SetPropertySwitch("Exposure", "Auto", 0)
                ## "0" is off, "1" is on.
        
                ExposureTime=[0]
                Camera.GetPropertyAbsoluteValue("Exposure","Value",ExposureTime)
                print("Exposure time abs: ", ExposureTime[0])
        
                #------------------------------------------------------------------------------------------
                # Set an absolute exposure time, given in fractions of seconds. 0.0303 is 1/30 second:
                #------------------------------------------------------------------------------------------
                Camera.SetPropertyAbsoluteValue("Exposure","Value",0.0303)
        
                #-----------------------
                # Set Camera gain : auto
                #-----------------------
                Set_Camera_gain(Camera, 0)
        
                #------------------------------------------------------------------------------------------
                # Start the live video stream, but show no own live video window. We will use OpenCV for this.
                #------------------------------------------------------------------------------------------
                Camera.StartLive(1)
        
                return Camera
            else:
                system_log.error("No device selected")
                return False

TIS-Stefan, 2021-07-06:
Hello
Your code looks very fine to me.

> After init camera, we call Getimage() to get the image data, but sometimes we get NULL point when we call Getimage.getimagePtr().

The code of getimagePtr() is:
```
unsigned char* AC IC_GetImagePtr( HGRABBER hGrabber )
{
	CGrabberHandle* hTmp = (CGrabberHandle*)hGrabber;
	unsigned char *pBytes = NULL;
	if( hTmp )
	{
		if( hTmp->pGrabber->isDevValid() )
		{
			pBytes = hTmp->m_pSink->getLastAcqMemBuffer()->getPtr();
		}
	}
	return pBytes;
}
```
That means, you get NULL returned only, if the camera becomes invalid. It is either not opened or it was somehow disconnected.
The tisgrabber.dll has the function
```
int AC IC_IsDevValid( HGRABBER hGrabber)
{
	int iResult = IC_ERROR;

	CGrabberHandle* hTmp = (CGrabberHandle*)hGrabber;
	if( hTmp )
	{
		if( hTmp->pGrabber != NULL  )
		{
			if(hTmp->pGrabber->isDevValid() ) iResult = IC_SUCCESS;
		}
	}
	return iResult;
}
```
That function can be used to check, whether the camera is still valid. It is in the tisgrabber.py as "IsDevValid" function. So you can check, whether the camera is valid before you grab an image.

Stefan

adamtsai66, 2021-07-07:
Thank you for the response.
We will try to check camera is valid(IsDevValid) before grab an image.

adamtsai66, 2021-07-28:
Thank you for your help. Updated the lastest result.
We can't duplicated the issue after we changed the camera driver from Cam33U_setup_4.5.0.1354 to Cam33U_setup_5.0.1.1608.

Do you know why it is related to camera driver?

TIS-Stefan, 2021-07-29:
Hello

> Do you know why it is related to camera driver?

Unfortunately not. I am sorry. There have been many changes and optimizations implemented between both versions.

Stefan

closed 2021-07-29 by TIS-Stefan

adamtsai66, 2021-07-29:
Understood, I still hereby give my infinite thanks for you.
Have a good day :)