TheImagingSource/IC3.5-Imaging-Control-Samples #34: Windows Error 0xe06d7363 (WinError -529697949) when setting gain (Python sample code on Windows)

gwappa, 2021-07-16:
Hi,

Thanks for providing the sample code like this!
I am trying to run `tis-OpenCV.py`, but the script keeps shutting down at line 79 (Python raises an OSError, with the error code as shown above):

```python
Camera.SetPropertyValue("Gain", "Value", 1)
```

When I comment the line out, everything else works out as expected, up to the point when I hit Ctrl-C.

Do you have any idea about what is going on?

In case, below is the description of my environment:

- Windows 10 (probably the latest build as of July 2021)
- Anaconda
  - Python 3.8.10
  - NumPy 1.20.3
  - OpenCV 4.0.1
- IC-Imaging-Control-Samples bb2368
- Camera: DMK 33UP1300 (Driver version: 5.0.1.1607), through a USB3 cable

Thanks,
   Keisuke

TIS-Stefan, 2021-07-16:
The gain minimum value of the DMK 33UP1300 is 144. Not 1. 
Please try 144.

gwappa, 2021-07-16:
Oh, I see. The line worked after I change the value to 144.
Is there a way to check the valid range of values, either programmatically or on the manual?

Thanks a lot!

TIS-Stefan, 2021-07-16:
Yes, there should be a range or a minimum function. 
At least, it is in the DLL. I am not sure, whether it is imported in Pyhton. (That is always double work,)

TIS-Stefan, 2021-07-16:
I see, there are no property range / min / max functions imported in tisgrabber. py. 
The function to be imported from the DLL is  IC_GetPropertyValueRange. See "tisgrabber.h" for a short documentation of this function.

gwappa, 2021-07-16:
Yeah, now I see it myself. Well, it seems anyway doable to export it to python for my own purpose anyway.

Everything becomes clear by now. Thanks again for your response!

closed 2021-07-16 by TIS-Stefan