TheImagingSource/IC3.5-Imaging-Control-Samples #39: Implementing callback function in python and return data in main loop?

nexus1203, 2021-11-03:
I have been trying to implement the callback function and return the captured image in the main loop, however the examples doesn't show how to do that. I am following this example. Some help will be very appreciated as I am trying to implement the hardware trigger function.  Or is there is another way to know if the Frame is ready from the main loop?

```

import ctypes as C
import tis_utils.tisgrabber as IC
import time as time
import cv2 as cv2
import numpy as np

import sys

def s(strin):
    if sys.version[0] == "2":
        return strin
    if type(strin) == "byte":
        return strin
    return strin.encode("utf-8")

class CallbackUserdata(C.Structure):
    """ Example for user data passed to the callback function. """
    def __init__(self):
        self.width = 0
        self.height = 0
        self.iBitsPerPixel = 0
        self.buffer_size = 0
        self.oldbrightness = 0
    

def Callback(hGrabber, pBuffer, framenumber, pData):
    """ This is an example callback function for image processig  with 
        opencv. The image data in pBuffer is converted into a cv Matrix
        and with cv.mean() the average brightness of the image is
        measuered.

    :param: hGrabber: This is the real pointer to the grabber object.
    :param: pBuffer : Pointer to the first pixel's first byte
    :param: framenumber : Number of the frame since the stream started
    :param: pData : Pointer to additional user data structure
    """
    print("yes")
    if pData.buffer_size > 0:
        image = C.cast(pBuffer, C.POINTER(C.c_ubyte * pData.buffer_size))

        cvMat = np.ndarray(buffer = image.contents,
                        dtype = np.uint8,
                        shape = (pData.height,
                                pData.width,
                                pData.iBitsPerPixel))
        cv2.imshow('image', cvMat)
        cv2.waitKey(1)
        brightness = cv2.mean(cvMat)
        b = int( brightness[0] )
        if b != pData.oldbrightness:
            print( b)
            pData.oldbrightness = b
        print("ok in")
        

# Create the function pointer.
Callbackfunc = IC.TIS_GrabberDLL.FRAMEREADYCALLBACK(Callback)

ImageDescription = CallbackUserdata()    

# Create the camera object.
Camera = IC.TIS_CAM()
Devices = Camera.GetDevices()

for i in range(len( Devices )):
    print( str(i) + " : " + str(Devices[i]))
# Open a camera. 
# Camera.openVideoCaptureDevice("DFx 72BUC02")
Camera.open(Devices[0].decode('utf-8'))
Camera.SetVideoFormat("Y800 (640x480)")
Camera.SetFrameRate(30.0)
# Noe pass the function pointer and our user data to the library.
Camera.SetFrameReadyCallback(Callbackfunc, ImageDescription )
# Handle each incoming frame automatically.
Camera.SetContinuousMode(0)

# Start live video.
Camera.StartLive(0)    

Camera.SetPropertySwitch("Exposure","Auto",0)
Camera.SetPropertyAbsoluteValue("Exposure","Value",1/30)

Camera.SetPropertySwitch("Trigger","Enable",1) # enable hardware trigger!

Camera.SetPropertySwitch("Gain","Auto",0)
Camera.SetPropertyValue("Gain","Value",34)

Imageformat = Camera.GetImageDescription()[:3]
ImageDescription.width = Imageformat[0]
ImageDescription.height= Imageformat[1]
ImageDescription.iBitsPerPixel=Imageformat[2]//8
ImageDescription.buffer_size = ImageDescription.width * ImageDescription.height * ImageDescription.iBitsPerPixel

ExposureTime=[0]

try:
    while ( True ):
        
        #'''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''
        # How to return the image data from the callback function.?
        # or know if frame is ready, so I can call Camera.GetImage() to get the latest frame
        # maybe something like 
        if Camera.FrameReady():
            image = Camera.GetImage()

        #''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''
        time.sleep(0.1)
        
except KeyboardInterrupt:
    Camera.StopLive()   

```

TIS-Stefan, 2021-11-03:
Hello

Currently I am changing the approach of using tisgrabber, so I changed the tisgrabber.py.

Please download the file(s)
 - tisgrabber.zip
 from https://we.tl/t-n30hFUvEHj
The link is valid for 7 days.

Please see samples/40-qt-stereo.py, maybe it is, what you are asking for.
It shows, how to
- use qt5 widgets to display live video in
- select and configure two devices
- use callbacks and create an opencv mat
- restore devices from saved xml files
- capture images from both cameras and save them to hard disc.

There are more samples. The main trick is making the image part of the user data struct used in the callback.

The images are handled in the main loop in this case in the GUI thread.

I hope this helps.

nexus1203, 2021-11-04:
Thank you for the updated library. It was helpful, however I wonder what happen to Camera.open("camera unique name") function. Its really helpful to be able to open cameras using unique names in multi-camera setup.

TIS-Stefan, 2021-11-04:
The function is IC_OpenDevByUniqueName() please see the updated tisgrabber.chm in the tisgrabber directory of this reposity. I uploaded it yesterday.

> wouldn't it be much better to simply assign the camera to a class like this and send the acquired image to the main UI?
Yes. The tisgrabber.dll C wrapper is not object oriented, therefore, I did not implement, rather removed, a camera class. The problem is, if I change the tisgrabber.dll, I also have to change the Python classes. This is a little bit too much work overload for me. 
But you can implement that class. 

> example is not working

What does that mean in detail? Which error message do you receive?
I am sure, my stereo sample works, so you may missed something?

Stefan

nexus1203, 2021-11-05:
> example is not working
I am working on a multi-camera setup (4 cameras DFM 42BUC03-ML). I am trying to run the four CameraWorker class in the background thread and get the data out of the pyqt5 signal/slot method (see the "def run" method). Because the default structure of I have to assign the camera and CallbackUserdata to an object in CameraWorker class like following:

``` Callbackfunc =  ic.FRAMEREADYCALLBACK(Callback)
            self.CameraState = CallbackUserdata()  
            ic.IC_SetFrameReadyCallback(self.Camera, Callbackfunc, self.CameraState)
            CreateUserData(self.CameraState, self.Camera)
```
After calling the worker class in main thread, the __init__ method works alright without any errors. but calling the "run" method doesn't return the cvMat object as it should in the example. The stereo example works, I believe because global variables are used and no "class" is used. In a more specialized setting, I want to be able to control each camera and their functions individually and not rely on global variables.  In my code, no errors occur but also camera state never updates : 
```
    @QtCore.pyqtSlot()
    def run(self):
               
        """start the camera acquisition"""
        i = 0
        fps = []
        self.CameraState.getNextImage = 1
        
        while self.threadactive:
            i+=1
            print("while loop count {}".format(i)) # this counter works and update the value of i 
            if self.CameraState.getNextImage == 0:
                # the self.CameraState.getNextImage  does not update and stays same
                # CallBackUserData assigned to the callback never updates and doesn't shows any errors.
                # 
                # this should return (emit) the image data to the main UI but nothing happens.
                print('State Changed')
                image = self.CameraState.cvMat
                self.data = {"id":self.camera_id, "image":image}
                self.signals.result.emit(self.data)
                self.CameraState.getNextImage = 1
            
            QThread.msleep(1)

TIS-Stefan, 2021-11-05:
The stereo sample works, because the grabber objects are in the same thread and there is only one thread used.
In an own multithready application with many cameras I called Initlibrary() in the thread of a camera. It looks like this:

```Python
def camerathread( data ):
    '''
    For each camera a thread is created. The in data.configfile saved camera.xml file
    is used to open the camera.
    The main loop runs for ever.
    If a device lost happens, the main loop tries to reopen the camera.

    Image saving to the "files" subdir is done here on demand.
    
    The snapped image data is compressed to jpeg in memory and saved in the passed "data.lockedimage" (limg) object.
    '''
    global ic
    
    ic.IC_InitLibrary(0)
    
    while data.running == True :
        hGrabber = ic.IC_CreateGrabber()
        ic.IC_LoadDeviceStateFromFile(hGrabber,tis.T(data.configfile))

        if( ic.IC_IsDevValid(hGrabber)): 
            # Add callbacks for framerate (not used) and device lost
            cb = ic.FRAMEREADYCALLBACK(Callback)
            devlostfunc = ic.DEVICELOSTCALLBACK(deviceLostCallback)
            ic.IC_SetCallbacks(hGrabber,cb,None,devlostfunc, data )

            ic.IC_StartLive(hGrabber,0)
```
(Ignore parts of the comment, the code is from a webserver showing many cameras in a browser)
( I am not sure, whether that "gobal ic" is necessary. I am not a Python expert, I am a beginner rather)

This works fine so far. 

I wonder, why you run the cameras in own threads. Each grabber (camera) uses its own thread, which calls the callback functions within that thread. Thus it is not necessary to create a new thread for the camera thread that already exists.
Within the callback you can call self.signals.result.emit(self.data). I made that at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/Python%20NET/qt5-simple.py

You could create a list containing the grabber objects and additional data in your main thread, open and configure the cameras, set the callback function (which is implemented once only and used by all cameras) and send the signal with the additional data and the cv::Mat (which is part of the additional data). You do not need an extra loop for snapping images in this case. Shall we proceed like this? Please let me know. If yes, I create a sample for you based on your code above. 

If you want to stay with your worker threas, I would like to suggest, that I rewrite your code to changed tisgrabber.py 
I will use the IC_OpenDevByUniqueName() function instead of IC_LoadDeviceStateFromFile(), So you do not need to create a device state file (IC_SaveDeviceStatetoFile() ).

Please let me know, how to proceed.

Stefan

nexus1203, 2021-11-05:
>Within the callback you can call self.signals.result.emit(self.data). 

Yes this what I am trying to figure out how to do that.

> You could create a list containing the grabber objects and additional data in your main thread, open and configure the cameras, set the callback function (which is implemented once only and used by [...]

I think this method will be ok as long as the pyqtsignal can be emitted from callback and received in the main UI thread. And I can be able to differentiate which camera the images belong to (probably using unique names?). Thank you very much for your help.

TIS-Stefan, 2021-11-05:
Hello

Well, I will start now. Base will be the stereo sample (I like copy&paste). 

> can be able to differentiate which camera the images belong to (probably using unique names?). T

That can all be written in the user data. I would not suggest to use different signals for each camera, because without that, you can use as many cameras as you want without changing the code.

Please give me some time. (I will handle my easier support cases first, so I am more concentrated.)

Stefan

TIS-Stefan, 2021-11-05:
Hello

Here we go: https://github.com/TheImagingSource/IC-Imaging-Control-Samples/tree/master/Python/tisgrabber/samples
it is 41-qt-triggering.py. The name suggests, there is something with trigger, but I did not implement enabling of trigger (you can do in the properties....)

The sample does:
- count the number of connected cameras 
- create for each camera a select and properties menu entry
- create a video window and a brightness progress bar
- implements a camera class for
  -  select device
  - device properties
  -  start, stop
  - save device state to file
  -  restore from file
  - create callback data
  - set callback
  - create user data
  - still missing: device lost event handler
- callback class 
  -  create the cv mat
  - send a signal to main ui thread on new image 
- image processing in main UI thread, show result in progressbar
- restores last used devices at program start. 

I suppose, the sample does, what you want to do.

There is a warning regarding IC_SetHwnd and internal conversion from void* to int. Thus, the sample works with Python 3.8, but not with 3.10. I think, the second parameter must be casted implicitly to int. 

I must admit, that signaling was tricky. I think the __namespace__ thing was necessary.

Stefan

TIS-Stefan, 2021-11-08:
Hello

You are right. Its in line 95 of tisgrabber.py. It must be
   ic.FRAMEREADYCALLBACK = ctypes.CFUNCTYPE(ctypes.c_void_p, ctypes.POINTER(HGRABBER), ctypes.POINTER( ctypes.c_ubyte), ctypes.c_ulong, ctypes.py_object )

I forgot to push this file. (There are some more differences, since it also support device lost events.)
The IC_SetHWnd() argtypes issue for Python 3.10 is not solved yet.

The tisgrabber.py is updated in the repository now.

Stefan

nexus1203, 2021-11-08:
thank you very much, its working without any errors.

TIS-Stefan, 2021-11-08:
And does the sample do, what you have in mind?

nexus1203, 2021-11-08:
>And does the sample do, what you have in mind?

Not out of the box but I have modified it to work with my user inference. And new callback function returns the images in both continuous and hardware trigger mode. Which helps a lot. Thank you very much for the support.

closed 2021-11-08 by nexus1203