TheImagingSource/IC3.5-Imaging-Control-Samples #41: Unable to open multiple cameras simultaneously

waterfall4289, 2021-11-24:
Greetings,

I am not able to stream from 2 cameras simultaneously with tisgrabber.py

python version: 3.8.12
pywin32-ctypes  version: 0.2.0 

I modified the sample program 01-Live.py to load two cameras but when I run it, only one camera is streaming. 
The modified program is below:
`'''
This sample demonstrates, how to open a camera with the
built in dialog and show a live video stream.
It is changed for import of tisgrabber.py file.
Needed DLLs for 64 bit environment are
- tisgrabber_x64.dll
- TIS_UDSHL11_x64.dll
- tisgrabber.py
'''
import ctypes
import tisgrabber as tis

ic = ctypes.cdll.LoadLibrary("./tisgrabber_x64.dll")
tis.declareFunctions(ic)

ic.IC_InitLibrary(0)

hGrabber1 = ic.IC_LoadDeviceStateFromFile(None,"camera0.xml".encode("utf-8"))
hGrabber2 = ic.IC_LoadDeviceStateFromFile(None,"camera1.xml".encode("utf-8"))

if( ic.IC_IsDevValid(hGrabber1)):
    print("camera1 is valid")
else:
    print("camera1 is not valid")

if( ic.IC_IsDevValid(hGrabber2)):
    print("camera2 is valid")
else:
    print("camera2 is not valid")    

if (ic.IC_IsDevValid(hGrabber1) and (ic.IC_IsDevValid(hGrabber2))):
    ic.IC_StartLive(hGrabber1, 1)
    ic.IC_StartLive(hGrabber2, 1)
    ic.IC_MsgBox(tis.T("Click OK to stop"), tis.T("Simple Live Video"))
    ic.IC_StopLive(hGrabber1)
    ic.IC_StopLive(hGrabber2)    
else:
    ic.IC_MsgBox(tis.T("No device opened"), tis.T("Simple Live Video"))

ic.IC_ReleaseGrabber(hGrabber1)
`

I tried running the sample program 40-qt-stereo.py and got the same result (only 1 camera streaming). The error messages below printed when I run 40-qt-stereo.py. I'm not sure if it's related to the problem I am having.

> (py3_8_12) D:\src\repos\IC-Imaging-Control-Samples\Python\tisgrabber\samples>python 40-qt-stereo.py
> 1.0.0.0
> QWindowsContext: OleInitialize() failed:  "COM error 0xffffffff80010106 RPC_E_CHANGED_MODE (Unknown error 0x080010106)"
> 40-qt-stereo.py:266: DeprecationWarning: an integer is required (got type sip.voidptr).  Implicit conversion to integers using __int__ is deprecated, and may be removed in a future version of Python.
>   ic.IC_SetHWnd( LeftCamera,  LeftVideo.winId())
> 40-qt-stereo.py:267: DeprecationWarning: an integer is required (got type sip.voidptr).  Implicit conversion to integers using __int__ is deprecated, and may be removed in a future version of Python.
>   ic.IC_SetHWnd( RightCamera,  RightVideo.winId())

How can I stream from multiple cameras simultaneously successfully?
Appreciate any help you can give.
Thank you.

TIS-Stefan, 2021-11-24:
Hello 

The "QWindowsContext: OleInitialize() failed: "COM error 0xffffffff80010106 RPC_E_CHANGED_MODE (Unknown error 0x080010106)" is normal and can be ignored.
The warning is "40-qt-stereo.py:266: DeprecationWarning: an integer is required (got type sip.voidptr). Implicit conversion to  integers using int is deprecated, and may be removed in a future version of Python." is also expected and will lead to an error with Python 3.10.

Unfortunately I need some more information.
What does "can not stream" mean? 
Which camera models do you use?
How are the camera connected to the computer?
Do the cameras run once at a time?

Suggestion: Lower the frame frame rate. I guess into the blue, there is not enough bandwidth on your connection for two cameras with more or less default settings.
```
    ic.IC_SetVideoFormat(hGrabber1, "RGB32 (640x480)".encode("utf-8"))
    ic.IC_SetFrameRate(hGrabber1, ctypes.c_float( 30.0))

    ic.IC_SetVideoFormat(hGrabbe2r, "RGB32 (640x480)".encode("utf-8"))
    ic.IC_SetFrameRate(hGrabber2, ctypes.c_float( 30.0))
```
That should work even for USB 2.0. If you use monochrome instead of color cameras, you use "Y800" instead of "RGB32"
(RGB32 is calculated from BayerRaw data from the camera, therefore, the bandwidth allocation of RGB32 is the sam as Y800.)

Stefan

TIS-Stefan, 2021-11-24:
I forgot to mention (because I do not know, which camera models you use):
Did you install the correct driver from https://www.theimagingsource.com/support/downloads-for-windows/ ? If not, then the system may can not distinguish between cameras. 

Stefan

waterfall4289, 2021-11-25:
Hi Stefan,

I am using the camera model DMK 23GM021.
Installed driver version 3.7.1.4512
The cameras do work one at a time.
I have two cameras connected to a gigabit switch connected to a gigabit RJ45 to USB3 adaptor to PC.

Good guess on not enough bandwidth. I had the framerate set to 30FPS and the format set to Y16 1280x960. By lowering the frame rate and image format I was able to stream(and by that I mean ic.IC_StartLive(hGrabber1, 1) opens up a window and shows a live camera view).

Thank you.

I am having a different issue now. 
I have two cameras showing live view correctly with StartLive(hGrabber1,1) and StartLive(hGrabber2,1).
I expect calls to ic.IC_SnapImage(self.hGrabber1, 2000) and ic.IC_GetImagePtr(self.hGrabber1) to return frame from hGrabber1 only and calls to ic.IC_SnapImage(self.hGrabber2, 2000) and ic.IC_GetImagePtr(self.hGrabber2) to return frame from hGrabber2 only.  What I am seeing, however, is that calls to ic.IC_SnapImage(self.hGrabber1, 2000) and ic.IC_GetImagePtr(self.hGrabber1) sometimes return frame from hGrabber2 and vice versa. I tried wrapping these calls in mutex but that does not seem to help, the problem persists. How do I consistently snap images from camera 1 and camera 2 without the images being mixed together?

[cameras.txt](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/7599897/cameras.txt)
[bar_view.txt](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/7599898/bar_view.txt)

waterfall4289, 2021-11-26:
I used 41-qt-triggering.py as a guide and got my code to work with two cameras with callback function. Frames don't seem to get mixed up now. Maybe an example like 41-qt-triggering.py but with IC_SnapImage instead? Thanks.

TIS-Stefan, 2021-11-26:
Hello
SnapImage does somethig completely different, than the callback code. I think, you should stay with the triggering, you can use the software trigger instead of snapimage.

But.... the error you report should not happen. It must be something in the tisgrabber.dll, which did not show up before. Since I use this DLL in Python for Windows, it is used in much more use cases, so it might happen to show some errors now. I am sorry for the error. When I have some spare time, I will check. 

Stefan

waterfall4289, 2021-11-26:
I am fine with using triggering. Thanks again for the help!

closed 2021-11-26 by waterfall4289