TheImagingSource/IC3.5-Imaging-Control-Samples #42: Monochrome formats (Y800 and Y16) still returing 3 bytes per pixel

gmoneyphx, 2021-11-25:
Using the 32-trigger-callback.py application with a 33GP006 camera and driver version 3.7.1.4512 if you select Y800 or Y16 format types the image size description:

# Query the image description values
    ic.IC_GetImageDescription(hGrabber, Width, Height, BitsPerPixel,
                              colorformat)

    # Calculate the buffer size
    bpp = int(BitsPerPixel.value/8.0)

still returns 3 bytes per pixel which causes higher than expected memory use.  Is there a way to end up with 1 or 2 bytes (Y800 / Y16) as expected?

TIS-Stefan, 2021-11-26:
Hello

In "tisgrabber.py" you will find
```
class SinkFormats(Enum):
   Y800 = 0
   RGB24 = 1 
   RGB32 = 2
   UYVY = 3  
   Y16 = 4
```
It was planned to be used as paramters to IC_SetFormat(HGRABBER, Format), but I ran into a Python casting issue from enum to int. Therefore, you use for Y800
ic.IC_SetFormat(grabber, 0)
and for Y16
ic.IC_SetFormat(grabber, 4)
If you use Y16 as sink format, make sure the videoformat of the camera is Y16 too.

The IC_SetFormat must be called before you start the live video stream. In case you use a callback, it must be called before you set the callback to the Grabber.

I hope, this helps.
(I am sorry for not finishing the development in time.)

Stefan

gmoneyphx, 2021-11-26:
It was important to set the format AFTER loading the camera xml definition even though the xml contains:

`<videoformat>Y800 (1600x1200)</videoformat>`

This does not work:

```
hGrabber = ic.IC_CreateGrabber()
frameReadyCallbackfunc = ic.FRAMEREADYCALLBACK(frameReadyCallback)
userdata = CallbackUserdata()
ic.IC_SetFormat(hGrabber, 0)
ic.IC_LoadDeviceStateFromFile(hGrabber, tis.T("camera1.xml"))
```

This does work:

```
hGrabber = ic.IC_CreateGrabber()
frameReadyCallbackfunc = ic.FRAMEREADYCALLBACK(frameReadyCallback)
userdata = CallbackUserdata()
ic.IC_LoadDeviceStateFromFile(hGrabber, tis.T("camera1.xml"))
ic.IC_SetFormat(hGrabber, 0)
```

Thanks for assistance.

TIS-Stefan, 2021-11-26:
>It was important to set the format AFTER loading the camera xml definition even though the xml contains:
Ah ok. Maybe it is, because the grabber object is created new and the default sink format is set to RGB24 then. I have to admit, I am currently now sure about this (its too long ago, when I wrote the DLL)

Thank you for showing, how it works!

Stefan

closed 2021-11-28 by gmoneyphx

ktgw0316, 2023-12-05:
@TIS-Stefan 
> It was planned to be used as paramters to IC_SetFormat(HGRABBER, Format), but I ran into a Python casting issue from enum to int.

If my understanding is correct, using `ic.IC_SetFormat(grabber, tis.SinkFormats.Y800.value)` instead of `ic.IC_SetFormat(grabber, tis.SinkFormats.Y800)` will work.  Anyway, `SinkFormats` should have been subclass of `IntEnum`, not `Enum` in this case.

TIS-Stefan, 2023-12-05:
Hello

You are right. But the tisgrabber will not be continued, because with IC4 Release (shortly (hopefully)) we publish a new C DLL, which is more sophisticated. 

Stefan

ktgw0316, 2023-12-05:
@TIS-Stefan Wow, I'm looking forward to the new release!