TheImagingSource/IC3.5-Imaging-Control-Samples #43: Maximum number of GigE cameras supported on a single computer?

gmoneyphx, 2021-11-28:
Does the GigE driver have a fundamental limit in the number of simultaneous cameras that are supported?  I want to connect 24 cameras through a TRENDnet switch that has a 10GbE uplink to my PC.  Each camera is putting out about 370Mbps (24fps 1600x1200 Y800) for an aggregate bandwidth of 8.8Gbps.  

I have tested with 2 cameras each saving 12 copies of the frame to disk (to simulate 24 streams) and this works ok (gen 4 NVMe).  Bottom line is I have 48 cameras to connect and want to do it with the minimum number of computers possible (hopefully 2)

TIS-Stefan, 2021-11-29:
Hello

There is no fundamental limit regarding the number of cameras. But you should consider, what amount of data your computer can handle. If all cameras stream simultaneously, this could become somewhat narrow and you will encounter frame drops.
If the cameras, in particular 33G and 38G family, are triggered at the same point of time with low trigger frequency, then there should be now frame drop expected, because the driver will request missing frames again, until all cameras delivered their frame. 
However, it is a little bit try and error. I must admit, the maximum GigE Cameras I connected to a 10GB network was 10. And that became narrow at the end, so the last camera encountered frame drops. I run every camera with maximum resolution and frame rate. 
The switch has 10 GigE and one 10GBit connectors.

Stefan

gmoneyphx, 2021-11-29:
Thanks, I will let you know soon how many cameras can work simultaneously in this specific environment.

closed 2021-11-29 by gmoneyphx