TheImagingSource/IC3.5-Imaging-Control-Samples #45: Java example property errors

revoltex, 2022-01-28:
Hi

When trying to run java example with DFK AFU050-L34 I got an error:
`Exception in thread "AWT-EventQueue-0" java.lang.NoClassDefFoundError: Failed to query property Gain
	at ic.PropertyDouble.GetRangePropertyDouble(Native Method)
	at ic.PropertyDouble.<init>(PropertyDouble.java:11)
	at ic.Grabber.getProperties(Native Method)`

Also it is not possible to change most of the properties (gain, exposure in GUI).

Is it possible that parameters for this camera are integer not double?
By the way, is there source code for ic.jar and JNI dll are available?

TIS-Stefan, 2022-01-28:
Hello 

Does the same happen with IC Capture from https://www.theimagingsource.com/support/downloads-for-windows/end-user-software/iccapture/?
Is the correct driver from https://www.theimagingsource.com/support/downloads-for-windows/device-drivers/icwdmuvccamtis/ installed? (I think so, otherwise you wont see the camera at all.)

I am sorry for asking the standard supporter questions first.

Can you send a screenshot?

The source code for the jar is not available. Also it is project with very low priority, because I have less than 1 questions for JAVA per year.

Stefan

revoltex, 2022-01-31:
[image]
[image]
[image]

IC Capture is working without problems. The latest drivers (v2.9.8) are installed.

The first screen is after running app, the second one is after selecting camera from dialog, and the final one is after we tried to change gain property.

TIS-Stefan, 2022-01-31:
The issue is caused by the data type of gain. The sample uses floating point / Absolute Values, while the camera supports long / Range only. Therefore, remove all the code regarding "GainArray" and pass the long values 0 to 1023 to

_grabber.PropertySet("Gain",GainSlider.getValue());

I am sorry for not had having the property value interfaces of the AFU050 in mind.
I hope, this helps!

Stefan

closed 2023-02-22 by TIS-Stefan