TheImagingSource/IC3.5-Imaging-Control-Samples #48: ctypes.ArgumentError: Expected LP_HGRABBER instance instead of LP_HGRABBER

vandoitruong1997, 2022-04-05:
Dear sir,

First, I ran an example 41 without error, however if I close that application and run the program gain, I got a " ctypes.ArgumentError " as below figure:
[image]

Current solution: restart the kernel every time when running the program.
However, it's quite inconvenient,  could you please give me some suggestions to fix this problem? 
Thank you so much!

vandoitruong1997, 2022-04-13:
Thank you for your reply!
I added your suggestion however it also occurred that problem.

I tried to add this line into the frameReadyCallback function:
print('thread index:', threading.get_ident(), "\n")
After exiting a program, Camera threads still exist and print out the thread index.
I think the IC_StopLive function could be a reason because it can not release camera.

TIS-Stefan, 2022-04-13:
Hello

At end of your program you should call
`ic.IC_ReleaseGrabber(hGrabber)`

Stefan

closed 2023-02-22 by TIS-Stefan