TheImagingSource/IC3.5-Imaging-Control-Samples #49: Open cameras with MODEL NUMBER and/or SERIAL NUMBER

SwHaraday, 2022-04-06:
Hi,

Before, we can use MODEL NUMBER and/or SERIAL NUMBER to open cameras like below.

Camera = IC.TIS_CAM()
Camera.open("DFK 33UX287 12345678")
Camera.StartLive(0)
Camera.SnapImage()
im = Camera.GetImage()
im = cv2.flip(im, 0)

I could not find sample which opens camera with MODEL NUMBER and/or SERIAL NUMBER.
Could you please let me know how to do.

thank you in advance !

TIS-Stefan, 2022-04-06:
You use the outdated Python implementation. It is no longer maintained. Therefore, you can no find a sample. Please look at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/tree/master/Python/tisgrabber for the current version.  Look at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/tisgrabber/samples/02-open-manually.py in particular:
```
ic.IC_InitLibrary(0)
hGrabber = ic.IC_CreateGrabber()
ic.IC_OpenVideoCaptureDevice(hGrabber, tis.T("DFK Z30GP031"))
```

Stefan

SwHaraday, 2022-04-06:
danke! Stefan,
i need to use 4 same models at once. So it is really important whether it
is possible to open cameras with SERIAL NUMBER.
Please simply tell by Yes or No first.

SwHaraday, 2022-04-06:
> Yes.
>
> That is possible. Use the function "IC_OpenDevByUniqueName"
> ic.IC_OpenDevByUniqueName(hGrabber[0],"DFK 33UX287 12345678")
> ic.IC_OpenDevByUniqueName(hGrabber[1],"DFK 33UX287 12345679")
> See the tisgrabber.chm for a documentation of the DLL:
> [image: image]
> <https://user-images.githubusercontent.com/4242120/161980825-5e719cc3-3228-4967-b47f-f6893b2a42f1.png>
>
> Stefan

Hi Stefan !
Vielen dank ! Great to know.
I'll try tomorrow !

yuji harada

SwHaraday, 2022-04-07:
Hi,
I'm trting, but this is pain.
All I want to do is to get images of 4ms exposure in every 15ms and
feed them to OpenCV process.
My code works with outdated dlls.
Why do you eliminate simple and useful command like
img = Camera.GetImage() ??

thanks

TIS-Stefan, 2022-04-07:
> Why do you eliminate simple and useful command like

Because it was no longer maintainable. When changing something in the tisgrabber.dll, I needed to make two additionally changes in the Python wrapper. That is was too much work overload.  The biggest advantage is, that I do not need to declare every function exported by the tisgrabber.dll in particular. It can be used directly.

How to grab an image and convert it to an OpenCV cvMat is shown at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/tisgrabber/samples/11-image-processing.py. From this, you can write a simple function for image grabbing:

```Python
def snapimage(hGrabber):
    image = None
    if ic.IC_SnapImage(hGrabber, 2000) == tis.IC_SUCCESS:
        # Declare variables of image description
        Width = ctypes.c_long()
        Height = ctypes.c_long()
        BitsPerPixel = ctypes.c_int()
        colorformat = ctypes.c_int()

        # Query the values of image description
        ic.IC_GetImageDescription(hGrabber, Width, Height,
                                  BitsPerPixel, colorformat)

        # Calculate the buffer size
        bpp = int(BitsPerPixel.value / 8.0)
        buffer_size = Width.value * Height.value * BitsPerPixel.value

        # Get the image data
        imagePtr = ic.IC_GetImagePtr(hGrabber)

        imagedata = ctypes.cast(imagePtr,
                                ctypes.POINTER(ctypes.c_ubyte *
                                               buffer_size))

        # Create the numpy array
        image = np.ndarray(buffer=imagedata.contents,
                           dtype=np.uint8,
                           shape=(Height.value,
                                  Width.value,
                                  bpp))
    return image                                      
```

You can enumerate and handle your four cameras as follows:

```Python
ic.IC_InitLibrary(0)
devicecount = ic.IC_GetDeviceCount()
grabbers = []
for i in range(0, devicecount):
    print("Device {}".format(tis.D(ic.IC_GetDevice(i))))
    uniquename = tis.D(ic.IC_GetUniqueNamefromList(i))
    print("Unique Name : {}".format(uniquename))

    # Uncomment this section, if you want to open all
    # video capture devices and show their live streams.

    g = ic.IC_CreateGrabber()
    ic.IC_OpenDevByUniqueName(g, tis.T(uniquename))
    grabbers.append(g)
```
Now you have a list containing objects for all connected cameras. Next step is configuration:
```Python
for grabber in grabbers:
    if(ic.IC_IsDevValid(grabber)):
        exposureauto = ctypes.c_long()
        #Gain
        ic.IC_SetPropertySwitch(grabber, tis.T("Gain"), tis.T("Auto"), 0)
        ic.IC_SetPropertyAbsoluteValue(grabber, tis.T("Gain"), tis.T("Value"),
                                   ctypes.c_float(0.0))
        #Exposure
        ic.IC_SetPropertySwitch(grabber, tis.T("Exposure"), tis.T("Auto"), 0)
        ic.IC_SetPropertyAbsoluteValue(grabber, tis.T("Exposure"), tis.T("Value"),
                                   ctypes.c_float(0.004))
        ic.IC_SetVideoFormat(grabber, tis.T("RGB32 (720x540)" ))
        ic.IC_SetFrameRate(grabber, ctypes.c_float(30.0))                                   
        # pass 0 instead of 1, if no video window should be visible
        ic.IC_StartLive(grabber, 1)
```
Snapping the images would be
```Python
mat0 = snapimage(grabbers[0])
mat1 = snapimage(grabbers[1])
mat2 = snapimage(grabbers[2])
mat3 = snapimage(grabbers[3])
```
I am not sure, whether this last one could be done in a loop too, because I am not that Python expert. 

At the end of your program you have
```Python
for grabber in grabbers:
    if(ic.IC_IsDevValid(grabber)):
        ic.IC_StopLive(grabber)
        ic.IC_ReleaseGrabber(grabber)
```

I hope, this reduces your pain a little bit. And I am very sorry to cause so much pain to you. 

Stefan

SwHaraday, 2022-04-12:
Hi,
They say no pain, no gain. Now I would say more pain, more gain.
I've read through tisgrabber.py, tisgrabber.h and TISGrabberGlobalDefs.h
again and realized that scripts are about the same. 
Pain is because of my poor knowledge.

Based on your suggestion, my modified code works almost as expected.
2 more questions come up.

1. How to set "Brightness" value in code ?

2. What is the best way to get images with very short exposure in interval which is much longer than exposure time ?

[image]

thank you !

yuji

TIS-Stefan, 2022-04-12:
Hello Yuji

Thank you very much for your comment!
Setting Brightness:
` ic.IC_SetPropertyRangeValue(grabber, tis.T("Brightness"), tis.T("Value"),ctypes.c_int(123))`

> What is the best way to get images with very short exposure in interval which is much longer than exposure time ?

The short answer is: use a callback. FrameReadyCallback. It is called every time, an image is received.

The long answer is the same. The FrameReadyCallback is called by the Grabber object. Each grabber object has its own thread, therefore, the FrameReadyCallbacks are running in own threads, so they are called in parallel for each of your four cameras. 

Make sure, the image processing does not last long, that 1/framerate seconds. 

The callback is explained at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/tisgrabber/documentation-source/images.rst#frameready-callback

A complete sample for callbacks with many cameras is available at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/tisgrabber/samples/41-qt-triggering.py As you can see, the callback function is define only once. All grabber objects use the same function in their thread. (That works!) (Triggering is not necessary, that can be disabled in that sample)

Stefan

SwHaraday, 2022-04-13:
Hi Stefan
Thank you very much for your support.
I guess my final question is far from this issue.
It would be better to close here and open another one.

bis morgen !
yuji

SwHaraday, 2022-04-13:
Hi Stefan
It says "AttributeError: function 'IC_PropertyRangeValue' not found"
[image]
Did I make something wrong ?
Also I could not find  'IC_PropertyRangeValue' in "tisgrabber.h".

Thank you.

TIS-Stefan, 2022-04-13:
My fault. The function is IC_SetPropertyValue():

[image]

SwHaraday, 2022-04-14:
Hi Stefan,
Now it works !
This will be on site next month.
In case of another issue, I will contact you.
I really appreciate your continuing cooperations !
Arigatou Gozaimasita !

yuji

TIS-Stefan, 2022-04-14:
Hello Yuji-san

Thank you very much. (I am sorry, my Japanese is totally insufficient)

Stefan

closed 2022-04-14 by TIS-Stefan