TheImagingSource/IC3.5-Imaging-Control-Samples #50: MediaStreamSink in IC WPF

KillSwitch607, 2022-05-03:
Hi Sir, 

I am wanting to record my camera output. Initially i tried to use MediaStreamSink as suggested in documentation. 

```
                MediaStreamSink aviSink = new MediaStreamSink();
                // We do not set aviSink.StreamContainer to record an AVI file,
                // as AVI is the default file format.
                // Set DV Video encoder as the codec.
                foreach (AviCompressor codec in AviCompressor.AviCompressors)
                {
                    if (codec.Name == "DV Video Encoder")
                    {
                        aviSink.Codec = codec;
                        break;
                    }
                }

                // Set a target file name.
                aviSink.Filename = "video.avi";
                // Set the sink.
                _ic.Sink = aviSink;
                // Start recording.
                _ic.LiveStart();
```
Apparently this option is NOT available in Live Mode. How are we supposed to record video while displaying the feed?

Thank you very much for your work on the project. Any help would be greatly appreciated.

TIS-Stefan, 2022-05-03:
Hello

Before assiging the new sink, you must stop the live stream with a call to _ic.LiveStop().

KillSwitch607, 2022-05-04:
Hi Mr. Stefan, 

Thank you for your response, I tried this and called LiveStop() as such, 
```
_ic.LiveStop();
 MediaStreamSink aviSink = new MediaStreamSink();
```
however,, while this does ensure a recording, I cannot see the camera feed while recording as it freezes the display as soon as I press the record button.

TIS-Stefan, 2022-05-04:
Hello

I suggest, you may look at the IC Imaging Control Sample in your "Documents/IC Imaging Control 3.5" directory. There you will see, how to capture an AVI file.
You can pause the video capture with https://www.theimagingsource.com/support/documentation/ic-imaging-control-net/prop_descBaseSink_SinkModeRunning.htm. So the AVI file is created after the sink is created. If StartLive is called, no frames are saved into the file, until the SinkModerRunnink property of the MediaStreamSink is changed.

Stefan

KillSwitch607, 2022-05-04:
I shall have a look at that. Is there any chance that there is a record program using WPF? The current examples are all WinForms. 

I appreciate your help.

TIS-Stefan, 2022-05-04:
I am sorry, but there is no recording program for WPF. 

I just saw in the caption you use WPF... I am so sorry, my answers are not helpful for you, because I did not understood your issue.
Now I understand. Using WPF you can not use the MediaStreamsink. But you can save the incoming frames on your own. The "Naudio" library is helpful for this. Currently I am out of office, so I do not have access to my sample library.  If you mind to remember me by next week, I show you, how to use the NAudio library for MP4 / h264 capture in C#

Stefan

KillSwitch607, 2022-05-04:
Hi Stefan

Thats totally fine, Ill have a look at the Naudio library. I would greatly appreciate your help, ill give you a message next week. Thank you once again. 

Regards
KillSwitch

KillSwitch607, 2022-05-09:
Hello Stefan, 

Just reaching out to you regarding the video recording. Is there any other way you would prefer to communicate or this place ok?

Kind Regards
KillSwitch

TIS-Stefan, 2022-05-09:
Hello

Please look at the sample of https://github.com/TheImagingSource/IC-Imaging-Control-Samples/issues/51#issuecomment-1121168067

Stefan

KillSwitch607, 2022-05-10:
Hi Stefan, 

Thanks for the sample. I really appreciate it. Does this method work for a WPF program?

Regards

KillSwitch607, 2022-05-10:
[ICWPF.zip](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/8660311/ICWPF.zip)

Hi Stefan, Ive uploaded a sample program of what Im working with.  Ive added some annotations in the code. I want the user to be able to record the video while the camera is still streaming. Im not sure how to do it from the winform app from #51. Do you have any insight into this? 

Regards
KillSwitch

TIS-Stefan, 2022-05-10:
Yes, this will work with WPF too. In the sample you use is an image callback, in which you receive the image. You can additionally forward this image to be saved in to the mp4 file, as shown in the sample in #51. Simply copy the parts you need from this sample.

Stefan

KillSwitch607, 2022-05-11:
Hi Stefan, 

Thank you for your response. Unfortunately I am still confused as to what to take from your sample. 
[image]

As you can see in the attached picture, I have a few buttons and a camera display. In this case, Im looking at the start recording and stop recording buttons. the image is displayed from a usercontrol I created. The select device button will trigger the capture rather than a start and stop button. Im not sure where to put what from #51, 

Please excuse my experience, I am quite new to C#. Your help is very much appreciated. Please let me know if you would like to see my source code if it may assist you. 

Kind Regards
KillSwitch

TIS-Stefan, 2022-05-11:
Hello

In the sample #51 in form1.cs you will find at the end:
```
 private void SaveBufferList(String FileName)
        {
            MFTestSharp.H264Writer writer;
            int BITRATE = 60 * 1000000;
            writer = new MFTestSharp.H264Writer(FileName, _sink.OutputFrameType, (int)icImagingControl1.DeviceFrameRate, BITRATE);
            writer.Begin();
            for (int i = 0; i < _bufferlist.Length; i++)
            {
                writer.Write(_bufferlist[i]);
            }
            writer.End();
        }
```
This function uses a list of IFrameQueueBuffers, that is written by the "writer" into a MP4 file.
That means for you: 
The "writer" will be an attribute (varialbe) of your class. You also need a flag, that indicates, you want to capture into a videofile

```
private  MFTestSharp.H264Writer _writer;
private bool _capturetovideo = false;
```

`You create and configure the writer when "Start Recording" was clicked. Also you set the _capturetovideo flag to true, so your class knows, that the frames are to be saved now:

```
            int BITRATE = 60 * 1000000;
            writer = new MFTestSharp.H264Writer(FileName, _sink.OutputFrameType, (int)icImagingControl1.DeviceFrameRate, BITRATE);
            writer.Begin();
          _capturetovideo  = true;
```
In the function `void DoUpdateImage(TIS.Imaging.IFrame buffer)` you add the buffer to the video file:
```
if( _capturetovideo )
    writer.Write(buffer );
```
When "Stop Recording" ist clicked, you have two lines of code:
```
_capturetovideo  = true;
_writer.End();
```
Since we are in a multithreaded situation here, the simple way to avoid conflicts is System.Threading.Sleep(200) between those lines. The correct way is different, but I do not want to confuse you.

You need the MFTestSharp.H264Writer stuff from the #51 sample.

Stefan

KillSwitch607, 2022-05-11:
Hi Stefan, 

Following your lead, this is what I have, I added H264Writer and the SinkListener to the code. 

```
        private void SaveBufferList(String FileName)
        {
            MFTestSharp.H264Writer writer;

            int BITRATE = 60 * 1000000;

            writer = new MFTestSharp.H264Writer(FileName, _sink.OutputFrameType, (int)_ic.DeviceFrameRate, BITRATE);
            writer.Begin();

            for (int i = 0; i < _bufferlist.Length; i++)
            {
                writer.Write(_bufferlist[i]);
            }
            writer.End();
        }
```

Then, the logic for the Start Recording Dialog, 
```
        private System.Windows.Forms.SaveFileDialog SaveFileDialog1;
        private void btnRecordCapture_Click(object sender, RoutedEventArgs e)
        {

            
            _capturetovideo = true;

            if (_ic.DeviceValid)
            {
                var dialog = new SaveFileDialog();
                dialog.FileName = "Capture";
                dialog.DefaultExt = "avi";
                var dialogresult = dialog.ShowDialog();
                if (dialogresult != true)
                {
                    SaveBufferList(SaveFileDialog1.FileName);
                }

            }

            if (!_ic.DeviceValid)
            {
                MessageBox.Show("Please start capture first.", "Error", MessageBoxButton.OK, MessageBoxImage.Warning);
            }
        }
```

Then the logic for Stop Recording dialog,

```
private void btnStopRecord_Click(object sender, RoutedEventArgs e)
        {
            if (_ic.DeviceValid)
            {
                _capturetovideo = true;
                _writer1.End();
            }

            if (!_ic.DeviceValid)
            {
                MessageBox.Show("Please start capture first.", "Error", MessageBoxButton.OK, MessageBoxImage.Warning);
            }
        }
```

Is this looking correct?

The thing regarding the `void DoUpdateImage(TIS.Imaging.IFrame buffer)` is that the function resides in the VideoWindow.cs file, which is a separate user control. I cannot put the following, without it throwing an error. 

```
if( _capturetovideo )
    writer.Write(buffer );
```

Thank you very much for your time. I really do appreciate it.

TIS-Stefan, 2022-05-11:
The "SaveBufferList" function is not needed in your code. It is to show, how simple this stuff is, only. 

Please read my previous answer carefully, because it tells you all, what you need to know!

Stefan

KillSwitch607, 2022-05-11:
Hi Stefan, 

Sadly I am still unable to get it to work. If you could look at my code and let me know where I've gone wrong, it would be of great help. I am desperate to get the video recording working.

[MainWindowCS.txt](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/8669357/MainWindowCS.txt)
[VideoWindowCS.txt](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/8669358/VideoWindowCS.txt)

these files are source files behind the main window and video window. Video window is edited to include just the basics. Im still quite lost :(.

TIS-Stefan, 2022-05-12:
Hello

Did you figure out, why your program can not save images to the video file? You did not set the _capturetovideo variable to true in "VideoWindow.cs". Therefore nothing is captured. 

I also should mention, you must call 
`
MediaFoundationApi.Startup();
`
once. This can be done in the constructor of your class.

Stefan

KillSwitch607, 2022-05-12:
Hi Stefan, 

Sadly not. I have set the `private bool _capturetovideo = true;` as you said, Also, I called `MediaFoundationApi.Startup` in the videoWindow constructor. The problem I have found is that in `private MFTestSharp.H264Writer _writer1;` in the VideoWindow.xaml.cs the _writer1 is never called. Hence, when I try to even run the camera, an error is thrown here. 

```
if (_capturetovideo)
            {
                _writer1.Write(buffer);
            }
```
Here, `_writer1` is null, because it was never initialized in the videoWindow. This is why it doesn't work. 

Regards

TIS-Stefan, 2022-05-12:
Hi

Here, _writer1 is null, because it was never initialized in the videoWindow. This is why it doesn't work.

So you must create the _writer1 as I showed yester.

Stefan

KillSwitch607, 2022-05-12:
Hi, 

I am sorry, I am very novice in programming (few weeks) hence I am having trouble with this. 

I created the `private MFTestSharp.H264Writer _writer1;`. By initialising, I assume you mean putting this in 

```
private void DoUpdateImage(IFrame buffer)

        if (_capturetovideo)
                       {
				int BITRATE = 60 * 1000000;
				_writer1 = new MFTestSharp.H264Writer(dialog.FileName, _sink.OutputFrameType, (int)_ic.DeviceFrameRate, BITRATE);
				_writer1.Write(buffer);
            }
```

Once again, my apologies for my inexperience.

TIS-Stefan, 2022-05-12:
Hi

Please analyze your code. It creates a new video file for each incoming image. Is this wanted?

Stefan

KillSwitch607, 2022-05-12:
No, I want to create a video file, when the button is pressed.

TIS-Stefan, 2022-05-12:
I thought so.

I am sorry, but it seems, you have not even basic knowledge about programming.  Please understand, that I can not teach you object oriented programming. 

Stefan