TheImagingSource/IC3.5-Imaging-Control-Samples #51: DisplayImageBuffer(IFrameQueueBuffer buffer) doesn't update ICImagingControl

5CoJL, 2022-05-05:
Hi,

I'm having a bit of an issue displaying saved frames one by one via DisplayImageBuffer, in .NET.

I have set up a FrameQueueSink alongisde a FrameQueueSinkListener. Whenever the listener has a frame queued in output, I save it in a IFrameQueueBuffer[] array to save the last 10 seconds of live imaging, and replay them when I hit a button.

However calling DisplayImageBuffer after calling ic.LiveStop() seems to never update the display (as in the last frame before the call to ic.LiveStop() is frozen).

I have set the ic.LiveDisplay to false before attempting to DisplayImageBuffer.

I understand I might be completely misunderstanding the tools at my disposal, but I have gone through the entire documentation as well as the samples and fall short of understanding my problem.

Thank you for any insight.

TIS-Stefan, 2022-05-06:
Hello

I have a complete sample that shows, what you want to do. Currently I am not in the office, therefore, I do not have access to my sample repository. I would like you to repeat your request on Monday or Tuesday, so I can provide better help.

Stefan

5CoJL, 2022-05-09:
Hello Stefan,

I reach out to you as suggested for the potential sample you mentioned ?

Best, Julien

5CoJL, 2022-05-09:
For the moment, I am using a workaround where I save my entire buffer as JPEGs, and display the images in a PictureBox.

I think I don't understand how the DisplayImageBuffer function works. I thought it would simply display whichever buffer I'd send the function, but apparently not.

TIS-Stefan, 2022-05-09:
Hello

[Capture before and after event.zip](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/8652530/Capture.before.and.after.event.zip)

You may try this sample. It displays the frame buffer queue in a picture box and allows to save the images into an mp4 file using the NAudio library.

No JPEG conversion necessary

Stefan

5CoJL, 2022-05-09:
Alright, thank you for this sample.

I managed to find another way to get what i was looking for with Ic Imaging Control.

However I still fail to understand how the DisplayImageBuffer function is supposed to work. Based on the documentation alone, I could not make it work the way it is described.

TIS-Stefan, 2022-05-09:
Hello

please look at the sample 
"Documents\IC Imaging Control 3.5\samples\C#\Display Buffer\Display Buffer\"

It passes a callback function "ShowBuffer" to the sink:
```
icImagingControl1.Sink = new FrameQueueSink(ShowBuffer, MediaSubtypes.RGB32, 5);
```

The display is:
```
        private FrameQueuedResult ShowBuffer( IFrameQueueBuffer buffer )
        {
            try
            {
                icImagingControl1.DisplayImageBuffer( buffer );
            }
            catch( Exception ex )
            {
                System.Diagnostics.Trace.WriteLine( ex.Message );
            }
            return FrameQueuedResult.ReQueue;
        }
```

5CoJL, 2022-05-23:
Hi, sorry for the delayed answer.

Indeed, this example works. But by using a FrameQueueListener to instanciate the FrameQueueSink, as opposed to the FrameQueuedResult of this example, then I could not get any display by calling DisplayImageBuffer.

closed 2023-02-22 by TIS-Stefan