TheImagingSource/IC3.5-Imaging-Control-Samples #54: Capturing to User-allocated Memory Buffers Example

westin123, 2022-08-05:
Hi, do you have any Python examples showing what is in the Capturing to User-allocated Memory Buffers documentation?
I'm having a hard time figuring out a way to make a FrameQueueBuffer object.

Thanks,

Westin

TIS-Kevin, 2022-08-08:
Dear Westin,

regarding our IC Imaging Control .NET Documentation there are multiple ways to create an instance of IFrameQueueBuffer.
You can find that here:
https://www.theimagingsource.com/support/documentation/ic-imaging-control-cpp/FrameQueueBuffer.htm

The Sample you are looking for can be found here:
https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/Python%20NET/qt5-mediastreamsink.py

Kevin

westin123, 2022-08-09:
Kevin,

Thanks for the information. I have been using the tisgrabber library and now I'm wondering which I should be using. Is this possible with that library?

Thanks,

Westin

TIS-Kevin, 2022-08-10:
Dear Westin,

the tisgrabber dll is based on our IC Imaging Control. So yes you can work with the tisgrabber library. I noticed that i linked the wrong sample last time, because the sample just shows how to create a FrameBuffer but the sample  "Imageprocessing.py" in our Python.NET directory on github shows how to use the iframequeuebuffer: 
https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/dc84380fdb8bcd2c11503da45a13bb30b8ba4b5c/Python/Python%20NET/qt5-imageprocessing.py

westin123, 2022-10-07:
Hi, Kevin. I was away working on other things but I now have time to work on this again. I have not seen that functionality in the tisgrabber library and I'm not super familiar with how it pulls functions out of the dll. Can you show me what the python call would look like to do that with tisgrabber?

What is the purpose of having two libraries for Python with similar functionality? Ideally my code would work under both Windows and Linux, and it seems like that may make .net a less ideal option.

Thanks

westin123, 2022-10-10:
@TIS-Kevin , alternatively is there a way to poll and see if a new image is ready in the tisgrabber python library?

TIS-Stefan, 2022-10-10:
Hello

Polling is using "IC_SnapImage" and check, whether there is an error (no image) or success (there is a new image). Please see the sample at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/tisgrabber/samples/10-save-image.py:

```Python
            if ic.IC_SnapImage(hGrabber, 2000) == tis.IC_SUCCESS:
                ic.IC_SaveImage(hGrabber, tis.T("test.jpg"),
                                tis.ImageFileTypes['JPEG'], 90)
                print("Image saved.")
            else:
                print("No frame received in 2 seconds.")
```
You can use a shorter timeout than 2000 milliseconds. 

Also you can find a complete tutorial at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/tree/master/Python/tisgrabber in "documentation" folder. 

Stefan

TIS-Stefan, 2022-10-10:
Hello Westin

> Hi, do you have any Python examples showing what is in the Capturing to User-allocated Memory Buffers documentation?
> I'm having a hard time figuring out a way to make a FrameQueueBuffer object.

I understand you having a hard time with that. The FrameQueueBuffer is something of IC Imaging Control .NET control, available with Python.NET. I do not know, what you want to do, but it is more simple to use the "tisgrabber" approach instead of Python.NET. They function completely different way and when I was learning Python, I thought, using .NET is more simple, than importing a DLL with ctypes. Later I learned, that ctypes is very simple and powerful. Therefore two approaches.

You may let me know, what you want to achieve, then I may can give better help.

Stefan

westin123, 2022-10-10:
@TIS-Stefan, my goal is to implement a wrapper library for interfacing with this camera which has a function to get the latest frame from the camera without waiting for the next one. The first thing I tried was to use the IC_SnapImage image with a very low timeout but found that this doesn't work well. When I have a value here of 0 or 1, the video comes through very choppy. It isn't a consistent frame rate, but its probably getting an average of 1 or 2 new frames per second. I'm assuming frames are being dropped. The plan with this was to return the previous frame if I get an error return code from SnapImage indicating that there is no frame ready yet. Is there any reason you know of that frames would be dropped when using a small value

westin123, 2022-10-10:
[image]

TIS-Stefan, 2022-10-11:
Hi

I wonder, for what you need a wrapper library, since the tisgrabber.py is an importer / wrapper.
However, I see now step by step, what you need to do, even if the complete image of your task is still unknown to me. 

The behavior using IC_SnapImage, you see, is as expected. If you want to have every frame single, you use a callback. Please see the sample at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/tisgrabber/samples/20-Callback.py

But I think, this is also not what you need, because I am guessing, without knowing, what exactly is your task. I understand, you may not want to write that in public, therefore, you can make a contact at https://www.theimagingsource.com/company/contact/. 

If I know, what you want to do, I can try to provide sample code for your task.

Stefan

westin123, 2022-10-12:
@TIS-Stefan , I'm making a wrapper because I'm supporting multiply camera types and I need a common interface. I tried the callback method you posted too and it is working for me. I remember trying that a while back and it didn't work so well. I'm not sure what I would have done wrong. Everything seems fine now. Thanks for the help.

Westin

closed 2022-10-13 by TIS-Stefan