TheImagingSource/IC3.5-Imaging-Control-Samples #55: Try to reach the framerate limit of camera

KJ-Chiu, 2022-09-20:
Hi, I'm using DFK 33UX287 and trying to have stable 300 images per second.
With RGB32 (720x540), 300fps and FRAMEREADYCALLBACK, I do get 300 callback per second. However the buffer_size is empty for most of the time.
Here is my code in callback
```
Width = ctypes.c_long()
Height = ctypes.c_long()
BitsPerPixel = ctypes.c_int()
colorformat = ctypes.c_int()

# Query the image description values
ic.IC_GetImageDescription(hGrabber, Width, Height, BitsPerPixel, colorformat)

# Calculate the buffer size
bpp = int(BitsPerPixel.value / 8.0)
buffer_size = Width.value * Height.value * bpp

if buffer_size > 0:
    imagedata = ctypes.cast(pBuffer, ctypes.POINTER(ctypes.c_ubyte * buffer_size))

    image = np.ndarray(
        buffer=imagedata.contents,
        dtype=np.uint8,
        shape=(Height.value, Width.value, bpp),
    )

    cv2.imwrite("./grab/{}.jpg".format(framenumber), image)
```

After several test I only got about 80 to 100 images per second, even using UI Panel "IC Measure" still not able to reach the 300 frames goal.

My develop env is i7-10750H CPU with 24GB RAM notebook, I'm not sure the problem is due to my hardware or incorrect code logic.
Is there any suggestion to find out the problem?

CedMrnl, 2022-09-21:
Hi, 

I'm have the same issue with a DMK 33UX287 with video format set to Y800 (720x540).
I also tried using "IC Capture 2.5" and "IC Measure" but the framerate limit was never reached. 

I'm also interested in a solution.

BR

TIS-Stefan, 2022-09-21:
Hello
IC Capture: 

Make sure, the exposure time is shorter than the frame rate. I pretty sure, you are aware of this. 

IC Capture shows in the lower right corner in the status bar the frame delivery statistics. You may let me know the numbers:
Meaning of the four numbers is
1 Packet Loss during transmission
2 Driver Transform issue, e.g. computer to slow
3 DirectShow Graph issue, e.g. computer to slow
4 Unknown

Python programming:

> However the buffer_size is empty for most of the time.

If buffer_size is 0, then width or height or bpp  are 0. This is not to be expected. If there is no frame, the callback is simply not called. 

Some more hints:
If you use a desktop computer, make sure the camera is connected to the backside connectors. The front side connectors may are bad shielded and you have issues on the physical layer of USB.
Make sure, power saving is disabled on the USB controller and the PCIe bus. It is in the Windows Power Options.
Some computer have issues with connect of USB 3 controllers to the rest of this computers. Instead of expected 75fps you get 4 only. In this case, assembling a new USB 3,0 controller board in the computer is a solution. Unfortunately you can get USB 3 controllers with ASMEDIA chipsets only, which can cause trouble.

Unfortunately these frame drops are usually caused by the computer and often hard to narrow down. 

Stefan

KJ-Chiu, 2022-09-21:
> Make sure, the exposure time is shorter than the frame rate. I pretty sure, you are aware of this.

Yes, my exposure time is shorter than the frame rate.

> IC Capture shows in the lower right corner in the status bar the frame delivery statistics. You may let me know the numbers:
Meaning of the four numbers is
1 Packet Loss during transmission
2 Driver Transform issue, e.g. computer to slow
3 DirectShow Graph issue, e.g. computer to slow
4 Unknown

I got 2/0/1928/0 after 10 seconds images grabbing with  RGB64 (720x540) and 300fps setting.

> Python programming

About the programming side. Thanks for the hint! buffer size zero's problem just solved after changing a USB3.0 port which is directly connected to the motherboard. Now the only issue is facing the bottleneck of hardware.

So now, I know that the frame drops are mainly caused by my computer. Do you have the suggestion specification of computer to stable handle 300fps and 720x540 RGB64 image grabbing? It will be pretty helpful to have this info

TIS-Stefan, 2022-09-21:
Well, having the 3rd number going up is really rare. So there is an issue with your DirectShow graph. You may try to set all properties of the camera do "default" and disable WDR/Tonemapping. 

It is a first guess.

Stefan

KJ-Chiu, 2022-09-21:
I'm using the Win11 system with DirectX 12.

> You may try to set all properties of the camera do "default" and disable WDR/Tonemapping.

looks like it doesn't help :'(

Two more things not sure if useful:
1. When changing the format to YUY2(720 x 540), the status after 10 seconds 300fps Image Sequence will be 3/2260/0/0.
2. When saving video with RGB32 avi file with 300fps. 10 seconds video will capture only 1857 frames with four zero status bar.

TIS-Stefan, 2022-09-22:
Hello 

There is something on your computer, that slows it down. I am lost.

Stefan

KJ-Chiu, 2022-10-20:
After several attempts, I solved the problem with using multi thread in Python.
I create two thread to save the images from queue which is put by the camera trigger callback.
Although the frame drop still happened (drop less then 10 frames in one second), it is already much better than before.

Short conclusion, I think the reason is my 3.6GHz CPU can't handle callback and image saving at the same time with single thread.

Thanks for your help, Stefan!

closed 2022-10-20 by KJ-Chiu

TIS-Stefan, 2022-10-20:
Hello

Thank you for letting me know your multi threading solution with threads for capture and saving (to hard disc?).

[IC-Express](https://www.theimagingsource.com/support/downloads-for-windows/end-user-software/icexpress/) saves images into memory first and does nothing else. When memory is full or all images are captured, then they are saved as single images. 

Stefan

PVaddy13, 2023-05-16:
Hello there,

Does anyone have a sample code for the suggested multi-threading solution?

Thank you,
Chandra.