TheImagingSource/IC3.5-Imaging-Control-Samples #58: DFK37BUX287:WDR function through tisgrabber.py

SwHaraday, 2022-12-12:
Hi
I'm using DFK37BUX287 through tisgrabber.py on Windows10 + Python3.7.

Is it possible to use WDR function with code below ?

~~~
ic.IC_SetPropertySwitch(hGrabber, tis.T("TneMapping"), tis.T("Enable"), 1)
ic.IC_SetPropertySwitch(hGrabber, tis.T("ToneMapping"), tis.T("Auto"), 0)
ic.IC_SetPropertyAbsoluteValue(hGrabber, tis.T("ToneMapping"), tis.T("Intensity"), ctypes.c_float(0.5))
ic.IC_SetPropertyAbsoluteValue(hGrabber, tis.T("ToneMapping"), tis.T("GlobalBrightnessFactor"), ctypes.c_float(0.0))
~~~

TIS-Stefan, 2022-12-12:
Yes, I think so. I use Python 3.8.10.

I suppose, you encountered an issue, right? Which one?

Stefan

SwHaraday, 2022-12-13:
Hi Stefan

Wie geht's ?

Thank you for quick response. They did not work with "ToneMapping" and "GlobalBrightnessFactor".  Then found that "Tone Mapping" and "Global Brightness Factor" work.
I've read several issue which you mention to disable Tone Mapping (WDR). Does it make any harm on acquisition in case of (FPS=70 Exposure=0.004sec) ?

Our next target products have shiny surface and are under dark environment. As I told you before, images from 4 direction are needed to feed to object detection.
I'm trying to achieve image as bright as possible playing with camera setting, because we experienced that lighting without harmful reflection is quite difficult. 
It is fine if only "Gain" work fine to achieve acceptable (bright enough and no noise) quality. 
If not, I suppose I have to use WDR. Am I right ?

TIS-Stefan, 2022-12-13:
Hello 

I am sorry, but I do not have all correct property names in mind, thus, my help was bad. (I could have looked on my own.)
You should also set the "Intensity" of Tone Mapping. 
> I've read several issue which you mention to disable Tone Mapping (WDR). Does it make any harm on acquisition in case of (FPS=70 Exposure=0.004sec) ?

No, it should not, except you have a very weak computer and run into trouble. Tone Mapping is done by software in the driver. The Tone Mapping values to be used depend highly on the light and dynamic situation. If you have a very low dynamic, which means, without Tone Mapping there is nothing very dark and very bright in the image, then Tone Mapping does not make too much sense. In this situation Gamma is more helpful. But it is mainly try and error.

> Our next target products have shiny surface and are under dark environment. As I told you before, images from 4 direction are needed to feed to object detection.
> I'm trying to achieve image as [...]

Reflection is always hard. In case you have plastic material to examine, you may try a polarisation filter. 
You should use the RGB64 video format and Tone Mapping, so you can take a little advantage of 10bits per channel color depth. 
Are the objects moving? If not, you should consider using HDR. Take four images with different exposure times and merge them: https://docs.opencv.org/3.4/d2/df0/tutorial_py_hdr.html. That could reduce the size of the reflections. You may try the sample: [42-HDR.zip](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/10215878/42-HDR.zip)

If the objects are moving, it becomes more complicated and you can play around with camera properties only. 

Stefan

SwHaraday, 2022-12-14:
Hi Stefan

Thanks for many suggestions.
Don't worry about property names.

PC for next project has enough power with USB3 expansion (ZVA-IOI PCIe USB3.0 Quad Channel 4 Port). 

Actual image taken at site contains the object and ceiling light.

[image]

I'm little confused. Is WDR(Tone Mapping) not equal to HDR ?
The object is moving at ca. 100mm/sec.
I understand HDR is not suitable for moving object, but how about Tone Mapping ?

Anyway, tomorrow I'm planning to visit our factory and play with camera properties.

Yuji

SwHaraday, 2022-12-16:
Hi Stefan

Thank you for detailed explanation. I believe I understood.

Yesterday I played with Gamma, Gain and WDR. Now it is quite sure that quality images can be achieved.

[image]

During try, I noticed that camera sometimes returned All-White images, when program restarted. Problem solved by unplugging and plugging in the USB. Can you guess what is going on ?
I'm not sure about repeatabe condition. So I leave this issue open.

Yuji

TIS-Stefan, 2022-12-16:
Hello Yuji

White images are not to be expected, in case all automatics are turned off.  However, that is my first guess. 

Stefan

closed 2023-02-22 by TIS-Stefan

SwHaraday, 2023-11-03:
Hi Stefan,

Now this system is working nicely.
Unfortunately device lost callbacks doesn't work, so I take the way to pick 40x40 pixels from center of each cam image and compare them to previous ones.
And judge device lost by count.
This process is finished less than 1msec.

Thank you for your kind contribution.
Excuse me for the late notification.

Yuji

TIS-Stefan, 2023-11-03:
Hello Yuji

If Device Lost is not fired, then there is no device lost at all. Maybe it is only, that your computer interrupts the image data strem transfer, so you do not get complete images anymore .