TheImagingSource/IC3.5-Imaging-Control-Samples #59: Confusion about version of tisgrabber.dll

bleykauf, 2022-12-13:
I found that the `tisgrabber.*` files in the [python samples](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/tree/master/Python/tisgrabber/samples) is not the same as in the latest version on the [website](https://www.theimagingsource.com/support/downloads-for-windows/software-development-kits-sdks/tisgrabberdll/) which is 3.4.0.51.

I assume that the library in the samples folder is 3.5 but I did not find any version information in `tisgrabber.h`. 

Could you please clarify this and possibly update the C library on the website or point me to the correct download link?

bleykauf, 2022-12-13:
Thank you for the quick reply.

So I guess delivering the `.dll`s with custom software is the way to go here if I wanted to use version 3.5. How is the library licensed, i.e. is it permitted to deliver `tisgrabber.dll` for non-commercial open-source projects?

TIS-Stefan, 2022-12-13:
The IC 3.5 Dlls can not be linked to tisgrabber. 3.4 only. 

The tisgrabber.dll as well as IC Imaging Control can be distributed for free for commercial and non commercial projects. 

Stefan

closed 2023-02-22 by TIS-Stefan