TheImagingSource/IC3.5-Imaging-Control-Samples #60: Framesnapsink/Framequesink

Passi089, 2022-12-14:
Hey,
I Have one question. I#m trying to programm an numerical algorithm for auto exposure and therefore i need the array of the images. The problem is when the images are placed in a IFrameQueueBuffer, due to the fact that they wer snapped using snapsink.SnapSingle, I#m able to refer to this data but the function has a TimeOut of minimum 2seconds otherwise i get an error. That's unfortunately to slow for my application. 
Is there another possibility doing what i want with the FrameSnapSink. Or should i use the FrameQueueSink, is it's better there? Or do you have any other suggsetions for me?

Kind regards,
Pascal

TIS-Stefan, 2022-12-14:
A timeout of 2 seconds is long. Usually an image is available with 1/framerate second, at worst case (1/framerate)*2 seconds. Which frame rate did you set?
And, which exposure time?

Stefan

Passi089, 2022-12-14:
Thank you for the answer. You are right. It was the long exposure time i had. 
But is it normal that the exposure time has to be like ten times smaller than the TimeOut, because that was what i'm figured out now. If The exposure time is like 0.05s i have to take a TimeOut of at least 3-4 seconds so that python don#t give me an error. Is that normal or is there a way to improve that?

If I'm taking short exposure times, everything is fine. 

Pascal

TIS-Stefan, 2022-12-15:
Pascal

> But is it normal that the exposure time has to be like ten times smaller than the TimeOut

No. Can you please post or attach your script? I will have a look.

Stefan

closed 2023-02-22 by TIS-Stefan