TheImagingSource/IC3.5-Imaging-Control-Samples #63: How to operate multiple cameras at the same time get image through python

jimmy2359, 2023-02-24:
Is there an example to operate multiple cameras , I'm not sure how to do this.
Should I choose to use method snapImage or callback? In Issues I have saw smbody
use "IC_LoadDeviceStateFromFile(None, tis.T("device.xml"))" to make it. But I still not 
figure out how to do rest of part。

Thanks for any help!

TIS-Stefan, 2023-02-24:
How to get and process the images depends on what you want to do. But may be the sample at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/tisgrabber/samples/40-qt-stereo.py, which uses two cameras in parallel is helpful

The other sample at https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/tisgrabber/samples/41-qt-triggering.py shows, how to open as many as connected cameras in parallel. 

Both cameras use triggering for synchronization, which may is not really necessary in your task. 

Stefan

closed 2023-03-01 by TIS-Stefan

jimmy2359, 2023-03-02:
I finish my main camera triger funtion, thanks for your help!