TheImagingSource/IC3.5-Imaging-Control-Samples #66: Capture AVI program is not working

PVaddy13, 2023-05-12:
Hi,

I am using a single TIS camera, and I am a complete newbie to this. As a starter, I am trying to run the provided sample codes. While running the "16-capture-avi-file.py", I am getting a white screen instead of the video I wish to capture. Looking for suggestions on what could have gone wrong.

To provide more information, I ran the program without any changes. I selected the default value of 0.75 when the codec quality was asked. Here is the link to the code which I am referring to.
https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/tisgrabber/samples/16-capture-avi-file.py

Any help in this matter would be greatly appreciated.

Thank you.
Chandra.

TIS-Stefan, 2023-05-12:
Hello Chandra

it is hard to guess, what is happening. I suppose, the simple live sample works.

You may use a smaller video format, because the MJPEG Compressor is not very efficient and will fail on bigger video formats. Thus, try 640x480 @ 30 fps for a start.

Stefan

PVaddy13, 2023-05-12:
Hello Stefan,

Thank you very much for the quick reply.

1. I tried the simple live program "1-Live.py". It's working perfectly.
2. For the "16-capture-avi-file.py," I used the MJPEG compressor as it was given in the sample code. I will try to use the other formats and let you know in future comments how it goes.
3. I am trying to capture a video with 200 fps (the project demands it). I assume the capture time would be less with such high capturing rate. Also, I am using "IC capture" software to check the focus and exposure. So I think the exposure time might not be the issue. I might be wrong. Please correct me if I am wrong.

Thank you,
Chandra.

PVaddy13, 2023-05-12:
Hello Stefan,

I tried using other video-capturing formats, but none of them worked. Can you provide a sample code for capturing an "avi video" that works for you?

Thanks,
Chandra.

TIS-Stefan, 2023-05-15:
Hi Chandra,

the provided sample works for me. 
But: for 200 fps the MJPEG Compressor is far to slow.  

Please let me know, which camera model, video format, framerate (200fps) and exposure time you use. I will have a look then.

Thank you in advance

PVaddy13, 2023-05-15:
Hello Stefan,

Here are the properties of the camera I am using.
model: DFK 37AUX273
video format: avi
framerate: 200 fps
exposure time: 1/30 ms (default value)

Thank you,
Chandra.

PVaddy13, 2023-05-15:
Hello Stefan,

When I run the code, a video file with the assigned name is generated, but no images are added to it. The generated avi file size is only 64 KB. If the captured images are added to the video, I assume the file size should be larger (let's say in MB at least).

Thanks,
Chandra.

TIS-Stefan, 2023-05-15:
Chandra

> I did not know about the trigger issue until I asked the other question

Same to me, therefore, the other question enlightened us.

Is the speed in the video low? In the properties mentioned above, use either "Auto" for exposure or set an exposure time below 1/200 seconds. 

Stefan

TIS-Stefan, 2023-05-15:
​
    

			
In github beanwortet.

	
	

						

	
		
										
							On May 15, 2023 at 3:36 PM, TheImagingSource/IC-Imaging-Control-Samples wrote:
					
	
	
		
							
                    

	Hello Stefan,
When I run the code, a video file with the assigned name is generated, but no images are added to it. The generated avi file size is only 64 KB. If the captured images are added to the video, I assume the file size should be larger (let's say in MB at least).
Thanks,
Chandra.

                    
                
					
	

 

								

	
		
										
							On May 15, 2023 at 2:56 PM, TheImagingSource/IC-Imaging-Control-Samples wrote:
					
	
	
		
							
                    

	Hello Stefan,
Here are the properties of the camera I am using.
model: DFK 37AUX273
video format: avi
framerate: 200 fps
exposure time: 1/30 ms (default value)
Thank you,
Chandra.

                    
                
					
	

 

								

	
		
										
							On May 12, 2023 at 7:54 PM, TheImagingSource/IC-Imaging-Control-Samples wrote:
					
	
	
		
							
                    

	Hello Stefan,
I tried using other video-capturing formats, but none of them worked. Can you provide a sample code for capturing an "avi video" that works for you?
Thanks,
Chandra.

                    
                
					
	

 

								

	
		
										
							On May 12, 2023 at 6:58 PM, TheImagingSource/IC-Imaging-Control-Samples wrote:
					
	
	
		
							
                    

	Hello Stefan,
Thank you very much for the quick reply.
I tried the simple live program "1-Live.py". It's working perfectly.
For the "16-capture-avi-file.py," I used the MJPEG compressor as it was given in the sample code. I will try to use the other formats and let you know in future comments how it goes.
I am trying to capture a video with 200 fps (the project demands it). I assume the capture time would be less with such high capturing rate. Also, I am using "IC capture" software to check the focus and exposure. So I think the exposure time might not be the issue. I might be wrong. Please correct me if I am wrong.
Thank you,
Chandra.

                    
                
					
	

 

					
	
					
	Was this message helpful?
	  Yes  
	  It was OK  
	  No  

		
			

		View and manage this ticket online:
		https://theimagingsource.deskpro.com/tickets/105876
	

 
The Imaging Source · https://theimagingsource.deskpro.com/
    GAQENBW5YBQ8PCH6Y95

    ​

     

@media only screen {
  html {
    min-height: 100%;
    background: #f3f3f3;
  }
}

TIS-Stefan, 2023-05-15:
​
    

			
In github beanwortet.

	
	

						

	
		
										
							On May 15, 2023 at 3:58 PM, TheImagingSource/IC-Imaging-Control-Samples wrote:
					
	
	
		
							
                    

	Hello Stefan,
Thank you for your quick responses. Actually, the trigger was the issue. I did not disable the trigger in the properties. To be honest, I did not know about the trigger issue until I asked the other question. I am able to capture a video in avi format for now. But as you mentioned the speed is very low.
Do you have any suggestions on how I can enhance/improve the capturing speed?

                    
                
					
	

 

					
	
					
	Was this message helpful?
	  Yes  
	  It was OK  
	  No  

		
			

		View and manage this ticket online:
		https://theimagingsource.deskpro.com/tickets/105942
	

 
The Imaging Source · https://theimagingsource.deskpro.com/
    GASS726QBTN8WCCBT4R

    ​

     

@media only screen {
  html {
    min-height: 100%;
    background: #f3f3f3;
  }
}

PVaddy13, 2023-05-15:
Thank you very much for your input, Stefan. Have a good one. :)

closed 2023-05-15 by PVaddy13