TheImagingSource/IC3.5-Imaging-Control-Samples #68: Exception: A set operation on a VCDProperty object failed

russellfoster, 2023-07-12:
I'm using software trigger to capture 8 frames at a time, at an interval of approximately one minute in a continuous process (24 hours/7 days per week). I use the same windows form/control throughout, and the camera device state is loaded and LiveStart() is called only during startup.

Every now and again (approx. 1-2 times per week) I run into this exception when calling softwareTrigger.Push() on the VCDButtonProperty:

TIS.Imaging.ICException: A set operation on a VCDProperty object failed.
   at TIS.Imaging.VCDButtonPropertyImpl.Push()

After that, any subsequent pushes give the same error, even if I retry them after 10 second delays (up to 10 times). This occurs in the middle of the 8 frame capture (it may at the beginning too, I am unsure of that). If I recreate the window and camera form, everything works fine. Any idea why this may be occurring? Is there some way else to reset the camera without relaunching the form/control?

russellfoster, 2023-07-12:
I do not have that event handler wired up. I can see about doing so. The model is DFK 37BUX264.

TIS-Stefan, 2023-07-13:
I suggest to add the device lost event handler of IC Imaging Control and check, whether it is called. Then we know, what happened. 
If that happen, you can try to reconnect the camera. 
A possible (oldschool) implementation for device lost could be
```C#
        private void DeviceLost()
        {
            showError(String.Format("Device {0} Lost!",_Name) );
            _ic.LiveStop();
            _ic.Device = "";
            // Try to reconnect after a device lost
            _reconnecttimer.Enabled = true;
        }

        private void _ic_DeviceLost(object sender, ICImagingControl.DeviceLostEventArgs e)
        {
            BeginInvoke(new DeviceLostDelegate(ref DeviceLost));
        }
```
The "reconnecttimer" event handler is
```C#
        private void _reconnecttimer_Tick(object sender, EventArgs e)
        {
            _reconnecttimer.Enabled = false;
            // If open of device failed, restart the timer.
            _reconnecttimer.Enabled = !openDevice();
        }
```
The "openDevice()" function is
```C#
        private bool openDevice()
        {
            try
            {
                _ic.LoadDeviceStateFromFile(_DeviceStateFile, true);
                _ic.LiveStart();
                return true;
            }
            catch (Exception ex)
            {
                _reconnecttimer.Enabled = true;
            }
            return false;
        }
```
The program I copied the code from, uses a "device.xml" for device configuration. The _reconnecttimer.Intervall should be not less than 200 (ms).

I hope, this helps a little bit

Stefan

russellfoster, 2023-07-13:
OK will try it out, thanks!

russellfoster, 2023-07-19:
OK, so we got another failure last night. We did get the DeviceLost event about 200ms before the VCDProperty error.

So this device lost event - what would cause it? The docs state "This event is called if the video capture device is no longer valid; for example if it was unplugged.",  Is there any way to determine the cause via software? Does it signal an issue with internal hardware (Camera), external hardware (USB cable), internal software (Imaging Source driver, library), external software (Consumer code), or some other unknown issue, or combination of these?

TIS-Stefan, 2023-07-19:
Hello
it is hard to say, what causes a random disconnect. Usually, it is something on the used computer, which causes this. There is no way to find out by software, what happened. Could be power saving.... 
In order to narrow down your problem, I need some more information:

Camera 

 - Camera model  :

 - Serial number :

 - Version of camera drivers :

In case you use an USB 3.0 camera:

Use Windows Device Manager to query following:

 - USB 3.0 Chipset (Intel, Renesas ....) :

 - USB 3.0 controller driver manufacturer / version : 

Computer 

 - Operating system :

 - Computer model :

 - Computer manufacturer :

 - Operating system :

 - Virus scanner : 

Software

 - Used software (e.g. IC Capture 2.0) :

In case you have a programming problem:

- Used programming language:

In case you use GigE cameras:

- IP Configuration of the network controller :

- IP configuration of the camera :

- Is the camera directly or via a network connected :

- Camera is powered by 1: PoE or 2: power supply :

Error:

 - Error message : random disconnect

 - Steps to reproduce the error : 

Thank you in advance! However, I can not promise to see anything in that data.  

Stefan

russellfoster, 2023-07-21:
OK thanks!  I will get these early next week.

russellfoster, 2023-07-28:
Sorry for the delayed response. Here is the requested info:

Camera 

 - Camera model  : DFK 37BUX264

 - Serial number : 28220136

 - Version of camera drivers : 5.1.0.1719

In case you use an USB 3.0 camera: Yes

Use Windows Device Manager to query following:

 - USB 3.0 Chipset (Intel, Renesas ....) : Intel(R) USB 3.0 eXtensible Host Controller - 1.0 (Microsoft)

 - USB 3.0 controller driver manufacturer / version :  Microsoft 10.0.19041.3031 5/8/2023

Computer 

 - Operating system : Windows 10 Pro 10.0.19044 Build 19044

 - Computer model : SBC-330P

 - Computer manufacturer : ASRock

 - Operating system : See above

 - Virus scanner : Windows 10 built-in

Software

 - Used software (e.g. IC Capture 2.0) : ICImaging Control

In case you have a programming problem:

Used programming language: C#
In case you use GigE cameras: No

IP Configuration of the network controller : N/A

IP configuration of the camera : N/A

Is the camera directly or via a network connected : N/A

Camera is powered by 1: PoE or 2: power supply : N/A

Error:

 - Error message : random disconnect

 - Steps to reproduce the error : Capture 8 frames every minute using the ICImaging Control. Disconnect occurs approx 1-2 times per week.

TIS-Stefan, 2023-07-31:
Hello
Thank you very much for your answer. Luckily I do not see any issues with your configuration. 
Unfortunately the "random disconnect" is one of the worst issues you can get, because you do not know, what causes the issue. This can be something weird in the computer. It also can be a loose contact in the USB cable. Or it can be something external, like an electro magnetic pulse caused by a machine turned on. 
As first measurement: Check for all power savings which can turn off devices. In Windows Device Manager check the properties of the USB controller. Some of them have under "Advanced" tab the checkbox "... can turn off device for power saving".
Also check the power plan of your computer for PCI bus and USB controller settings. However, that is guessing and must not really help. 

Since the camera does not seem to physically disconnect and is available after Device Lost event without doing anything, I suggest to try the automatic reconnect, I already mentioned.

Stefan

russellfoster, 2023-08-02:
OK will do. Thanks!

closed 2023-08-02 by russellfoster