TheImagingSource/IC3.5-Imaging-Control-Samples #69: IC_StartLive fail

Ezreal105, 2023-07-26:
Hello,
I am opening the camera using IC_OpenDevByUniqueName, then setting the continuous capture mode using IC_SetContinuousMode. However, when I call IC_StartLive, it fails and returns a status code of 0. The model of my camera is DMM 37UX273-ML. It's worth mentioning that this issue is not consistent. I have three cameras, and sometimes all three are working fine, but occasionally one of them encounters this problem. I am using the same code to open all three cameras simultaneously. Could you please provide any suggestions or ideas on how to troubleshoot this issue? Thank you.

Ezreal105, 2023-07-26:
I have successfully called IC_SetVideoFormat('Y800 (1440x1080)') before IC_StartLive.

TIS-Stefan, 2023-07-26:
Hello
Please show me your code. Also make sure, the correct driver is installed, because Windows Device Manager sometimes uses the Windwos camera driver. Then you get video formats like "mono8". You can check that with our IC Capture. 

Stefan

Ezreal105, 2024-07-08:
I finally discovered that it was an issue with the driver version. When I upgraded the driver, the problem was resolved. Thanks again.

closed 2024-07-08 by TIS-Stefan