TheImagingSource/IC3.5-Imaging-Control-Samples #71: Get crashed on create MediaStreamSink object in C# in .net core 7

TheSnowfield, 2023-10-24:
My camera is 38UX304.
The demo **Image And AVI Capture** works well under .net fx 4.8.
But my project uses .net core 7,  therefore I tried to port 'TIS.Imaging.ICImagingControl35.dll' on my project and finally get crashed.

Honestly, I did nothing. To reproduce the issue I created a blank WinForm project and created a MediaStreamSink object after component initialization.
[image]

What happened? I'm sure the files are ready.
[image]

I tried to upgrade the demo project from .net fx 4.8 to .net core 7 directly, without moving files or changing the code. 
The result was fried my brain. I'm screaming how does it work?! ~~I believe there are some magical things stuck me here.~~

Any idea is looking forward. my best thx :)

TIS-Stefan, 2023-10-25:
Hello

I guess, there is no IC Imaging Control. If I am right:

Please create an IC Imaging Control before creating a Mediastream sink. I guess, the error happens, because the COM library is not initialised
```C#
public partial class Form1 : Form
{
	private MediaStreamSink _avisink;
	private ICImagingControl _ic = new ICImagingControl()
	
	public Form1()
	{
		InitializeComonent();
		_avisink = new MediaStreamSink();	
	}
```
Maybe posting a minimal reproducible project shows more to me. (https://stackoverflow.com/help/minimal-reproducible-example)

Stefan

TheSnowfield, 2023-10-25:
> Hello
> 
> I guess, there is no IC Imaging Control. If I am right:
> 
> Please create an IC Imaging Control before creating a Mediastream sink. I guess, the error happens, because the COM library is not initialised
> 
> ```cs
> public partial class Form1 : Form
> {
> 	private MediaStreamSink _avisink;
> 	private ICImagingControl _ic = new ICImagingControl()
> 	
> 	public Form1()
> 	{
> 		InitializeComonent();
> 		_avisink = new MediaStreamSink();	
> 	}
> ```
> 
> Maybe posting a minimal reproducible project shows more to me. (https://stackoverflow.com/help/minimal-reproducible-example)
> 
> Stefan

Thanks for your quick reply! 
I've tested your code, the exception was gone! 
I can't believe there's a connection between two independent classes.

closed 2023-10-25 by TheSnowfield