TheImagingSource/IC3.5-Imaging-Control-Samples #72: Web browser stream

rafyzg, 2023-12-19:
Hey TheImagingSource,

Thank you very much for your packages and the support.
I was wondering if you know a way I could video stream from the camera to the browser in windows?
I saw an official example from TIS but unfortunately it only supports linux.
https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/blob/master/python/webserver/tcam_webserver.py

Is it possible to implement it in windows?
Thanks.

TIS-Stefan, 2023-12-20:
Yes, it is possible:
[tisgrabber-imagewebserver.zip](https://github.com/TheImagingSource/IC-Imaging-Control-Samples/files/13724813/tisgrabber-imagewebserver.zip)
However, the sample is not very well, because it wont stop the image forwarding thread, when pressing Ctrl +  C.
Also it is not very efficient. 

You may think about using webRTC. https://webrtc.org/?hl=en. Maybe the driver of your camera must be limited to show only RGB24, because webRTC does not like the mono formats of your camera. I would show, how to do that, but I do not know, which camera model (GigE or USB) you use.

Stefan

rafyzg, 2023-12-20:
Thank you very much stefan.
I am using a gige camera.
I would really appreciate an example of how to do it with webrtc.

TIS-Stefan, 2023-12-20:
Hello
The samples and even working live video in the browser is on the webRTC page. There you can select the camera and show the video.
You can add following key to the registry in order to limit the available video formats to RGB32:
```
[HKEY_LOCAL_MACHINE\SOFTWARE\The Imaging Source Europe GmbH\GigECamDevServer\devices\_service_defaults]
"allowed_fcc_0"=dword:34424752
```
Stefan

rafyzg, 2023-12-24:
Hey Stefan,
Indeed the  WebRTC examples works great! 
I am wondering if you are aware of  a way to load device state with WebRTC? 
Basically I have a settings I am defining using ICCapture software, could I use this device settings when connecting with WebRTC?

Thank you.

TIS-Stefan, 2024-01-02:
Hi

A happy new year to you!

Good to know, webRTC works fine. 

> I am wondering if you are aware of a way to load device state with WebRTC?

I am sorry, but this webRTC is not a software from us, thus, we can not interfere too much. But you can load a previously, with IC Capture 2.5 exported Device-State xml file with a small Python script:

```Python
import ctypes
import tisgrabber as tis

ic = ctypes.cdll.LoadLibrary("./tisgrabber_x64.dll")
tis.declareFunctions(ic)
ic.IC_InitLibrary(0)
hGrabber = ic.IC_CreateGrabber()
ic.IC_LoadDeviceStateFromFile(self.hgrabber, tis.T("camera.xml"))
ic.IC_ReleaseGrabber(hGrabber)
```
This sets all properties except video format and frame rate, which is set again by webRTC.

(Working with Linux you could use the startup behavior properties of the camera. The restored settings are overwritten in Windows by the driver, which saves the last used properties in the registry and restores them.)

I hope, this helps. 
Stefan