TheImagingSource/IC3.5-Imaging-Control-Samples #73: Getting the exposure time

Passi089, 2024-01-22:
Hallo Stefan,

vielleciht kannst du mir weiterhelfen. In meinem Porgramm kann ich die device properties aufrufen und darüber auch die exposure time setzten. Dies ermöglicht mir die folgenden Zeilen:

def ShowProperties():
    if ic.DeviceValid is True:
        ic.ShowPropertyDialog()
        ic.SaveDeviceStateToFile("device.xml")

Nun möchte ich das die dabei gesetzte exposure time in einer python variable gespeichert wird. ich weiß jedoch nicht wie ich darauf zugreifen kann. Hast du eine Idee für mich? Danke schonmal.

TIS-Stefan, 2024-01-22:
Hallo

die PropertyMap Dokumentation ist unter https://www.theimagingsource.com/en-us/documentation/ic4python/technical-device-properties.html.

Die Eigenschaften werde mit
```Python
    grabber.device_property_map.set_value(ic4.PropId.EXPOSURE_AUTO, "Off")
    grabber.device_property_map.set_value(ic4.PropId.EXPOSURE_TIME, 1000.0)
```
gesetzt und mit 

```Python
    exptime = grabber.device_property_map.get_value_float(ic4.PropId.EXPOSURE_TIME)
```
gelesen werden. (Ich habe das nicht ausprobiert.)

Stefan

Passi089, 2024-01-22:
Thanks for the response. I'm using the simple version with creating a ic object. Do you know what i can do if i don't use a grabber? Or how to easy implement a grabber?

# Create the IC Imaging Control object.
    ic = TIS.Imaging.ICImagingControl()

    snapsink = TIS.Imaging.FrameSnapSink(TIS.Imaging.MediaSubtypes.Y800)
    ic.Sink = snapsink

    ic.LiveDisplay = False

    # Try to open the last used video capture device.
    try:
        ic.LoadDeviceStateFromFile("device.xml", True)
        if ic.DeviceValid is True:
            ic.LiveStart()
    except Exception as ex:
        print(ex)
        pass

TIS-Stefan, 2024-01-22:
Hallo

Entschuldigung, ich war schon bei IC Imaging Control 4. Die Lösung für tisgrabber.dll ist:

```Python
    exposure = ctypes.c_float()
    ic.IC_GetPropertyAbsoluteValue(hGrabber, tis.T("Exposure"), tis.T("Value"),
                                   exposure)
```
Das komplette Beispiel ist unter https://github.com/TheImagingSource/IC-Imaging-Control-Samples/blob/master/Python/tisgrabber/samples/05-properties.py