TheImagingSource/IC3.5-Imaging-Control-Samples #75: Ic imaging control 3.5

Passi089, 2024-03-12:
Hi i'm using ic imaging control 3.5 in python.  

I require a image using Frame = snapsink.SnapSingle(TimeSpan.FromSeconds(2)) then i use a pointer = TIS.Imaging.IFrame(Frame) and through the pointer i can then acesss the image array. 

I use this function in a threading module in order to aquire an image every time the loop is running. This works very well for low frequency of image aquirering, but if i want to use it very fast something like 90 fps at some point the GUI crashes.  I think it has to do something with the frame and how the images are saved and the memory is acessed. how can i do that in a better way, do you have any ideas?

Passi089, 2024-03-12:
I use this function:
    def get_snapped_image(self):

            Frame = snapsink.SnapSingle(TimeSpan.FromSeconds(2))

            pointer = TIS.Imaging.IFrame(Frame)
            pointer.GetIntPtr()

            imgcontent = C.cast(Frame.GetIntPtr().ToInt64(), C.POINTER(C.c_ubyte * Frame.FrameType.BufferSize))

            gray = np.ndarray(buffer=imgcontent.contents, dtype=np.uint8,
                              shape=(Frame.FrameType.Height, Frame.FrameType.Width))

           return gray

This is for the start of the programm:

if __name__ == "__main__":

    # Create the IC Imaging Control object.
    ic = TIS.Imaging.ICImagingControl()

    snapsink = TIS.Imaging.FrameSnapSink(TIS.Imaging.MediaSubtypes.Y800)
    ic.Sink = snapsink

    # Try to open the last used video capture device.
    try:
        ic.LoadDeviceStateFromFile("device.xml", True)
        if ic.DeviceValid is True:
            ic.LiveStart()
    except Exception as ex:
        print(ex)
        pass

then i use the workerthread of pyqt5 to run a long running task at the end the gray picture which is created is emitteted to the GUI and displayed there. In the simplest case just the picture with no calulcation or so one is snapped and emitted to the GUI. This can be done as fast in terms of fps as the camera settings are set. When i start the programm everything is fine, but after some time it crashes with the following code: -1073741819. And it seem to be that it is due to that no frame is caputerd in this case. How can i prevent this due you have any idea? I want to have the function snapsingle for my case. So if it is possible to realise it stable with the FrameSnapSink it would be perfect.

TIS-Stefan, 2024-03-12:
Hi

Frame = snapsink.SnapSingle(TimeSpan.FromSeconds(2))

throws an exception on error. Also frame is null, (none) if no frame was snapped. I would like you to wrap this code in a try except construct, in order to avoid null pointer errors, which may is, what happens. 

Stefan