TheImagingSource/IC3.5-Imaging-Control-Samples #77: Python (tisgrabber): interaction with camera livestream IC_StartLive

brianavila72, 2024-04-09:
Hi,

I would like to know if it is somehow possible to interact with the camera livestream of IC_StartLive. I would like to know for example in which pixel position is my cursor on the image or draw a pointer in a specific pixel position on the stream.

TIS-Kevin, 2024-04-12:
Dear brianavila72,
yes you can.
For example you can use the MouseDown Event of the icImagingControl Object. With that you will receive a Point but this received Point is the location of your Mouse in the icImagingControl but not the exact pixel location of the Image. So you have to correct this Point regarding the Resolution of your Image.
The Calculation would be something like that:
X Location in image = X Location in IcImagingControl Ojbect * FrameWidth / LiveDisplayWidth
Y Location in image = Y Location in IcImagingControl Ojbect * FrameHeight / LiveDisplayHeight

brianavila72, 2024-04-12:
Dear Kevin, 
thank you for your reply. Unfortunately, I can't find any documentation on how to use the MouseDown Event, except this in Visual Basic [https://www.theimagingsource.com/en-us/documentation/icimagingcontrolactivex/event_descICImagingControl_MouseDown.htm](url), but it doesn't seem very useful to me. Sorry if this is a very basic question, but could you provide me with more information on this?

TIS-Stefan, 2024-04-15:
Hello 

The mouse events are usually provided by your GUI framework, e.g. QT. The window opened by tisgrabber has no message queue at all, so no mouse events are available. 

Stefan

brianavila72, 2024-04-18:
Hi,

I am using QT for my application, when I press a button on my GUI the livestream is shown with the window opened by tisgrabber (IC_StartLive). Do you mean that even though this window has no message queue, QT should still be able to retrieve the Mouse location?

Brian

TIS-Stefan, 2024-04-19:
Hi Brian
I think so, but I never tried that on my own.  QWidget  has mouse events, thus I think, it should work. 

In case, your project is new, you may consider using IC Imaging Control 4 Python (https://www.theimagingsource.com/en-us/documentation/ic4python/index.html) . Your QT code wont change. But the documentation of the display does not tell too much about mouse usage.